keirf/greaseweazle #1: Hang on reading track

solarmon, 2019-10-18:
Greaseweazle will randomly hang on reading from a PC drive.

Here is the traceback when python is broken out of using CTRL-C:

`

C:\Users\monideth\Downloads\Greaseweazle\Greaseweazle-v0.1>C:\Users\monideth\AppData\Local\Programs\Python\Python37\python gw.py read x-copy.scp COM14

Reading Track 18.0...Traceback (most recent call last):
  File "gw.py", line 345, in <module>
    main(sys.argv)
  File "gw.py", line 339, in main
    actions[args.action](args)
  File "gw.py", line 213, in read
    flux = read_flux(args.revs)
  File "gw.py", line 154, in read_flux
    x += ser.read(1)
  File "C:\Users\monideth\AppData\Local\Programs\Python\Python37\lib\site-packages\serial\serialwin32.py", line 290, in read
    True)
KeyboardInterrupt

`

Above is for a Python 3.7 64-bit environment, but have also tried in a Python 3.7 32-bit environment, on the same machine which is Windows 10 64-bit.

This is also after increasing the timeout in gw.py:

`
parser.add_argument("device", timeout="5")
`

solarmon, 2019-10-18:
This issue also occurs on an Amiga floppy drive (with SELECT on pin 10) and a Gotek drive (with SELECT on pin 10).

keirf, 2019-10-19:
The correct way to set a timeout is serial.Serial(args.device, timeout=5)

Ie you changed the wrong line.

solarmon, 2019-10-19:
Sorry, I wrote that whilst half asleep!

I meant:

`ser = serial.Serial(args.device,timeout=5)`

And yes, with that I can use CTRL-C to break out of the python script.

drdpj, 2019-10-21:
I'm also getting this behaviour :( -
`Reading Track 14.0...Traceback (most recent call last):
  File "C:\Users\djame\Downloads\Greaseweazle-v0.1\Greaseweazle-v0.1\gw.py", line 345, in <module>
    main(sys.argv)
  File "C:\Users\djame\Downloads\Greaseweazle-v0.1\Greaseweazle-v0.1\gw.py", line 339, in main
    actions[args.action](args)
  File "C:\Users\djame\Downloads\Greaseweazle-v0.1\Greaseweazle-v0.1\gw.py", line 213, in read
    flux = read_flux(args.revs)
  File "C:\Users\djame\Downloads\Greaseweazle-v0.1\Greaseweazle-v0.1\gw.py", line 154, in read_flux
    x += ser.read(1)
  File "C:\Users\djame\AppData\Local\Programs\Python\Python38\lib\site-packages\serial\serialwin32.py", line 286, in read
    result_ok = win32.GetOverlappedResult(
KeyboardInterrupt
^C`

also:
`C:\Users\djame\Downloads\Greaseweazle-v0.1\Greaseweazle-v0.1>gw.py read grannyG.scp COM13
Reading Track 58.0...Traceback (most recent call last):
  File "C:\Users\djame\Downloads\Greaseweazle-v0.1\Greaseweazle-v0.1\gw.py", line 345, in <module>
    main(sys.argv)
  File "C:\Users\djame\Downloads\Greaseweazle-v0.1\Greaseweazle-v0.1\gw.py", line 339, in main
    actions[args.action](args)
  File "C:\Users\djame\Downloads\Greaseweazle-v0.1\Greaseweazle-v0.1\gw.py", line 213, in read
    flux = read_flux(args.revs)
  File "C:\Users\djame\Downloads\Greaseweazle-v0.1\Greaseweazle-v0.1\gw.py", line 155, in read_flux
    x += ser.read(ser.in_waiting)
  File "C:\Users\djame\AppData\Local\Programs\Python\Python38\lib\site-packages\serial\serialwin32.py", line 257, in in_waiting
    raise SerialException("ClearCommError failed ({!r})".format(ctypes.WinError()))
serial.serialutil.SerialException: ClearCommError failed (PermissionError(13, 'The device does not recognize the command.', None, 22))`

keirf, 2019-10-22:
Cheers for the tracebacks!

drdpj, 2019-10-22:
I've got an STLINK so if I get the environment set up I should be able to do some proper debugging on the STM side (should it be required)....

keirf, 2019-10-29:
This is fixed in v0.2 to be released later today.

closed 2019-10-29 by keirf