keirf/greaseweazle #2: Will this RobotDYN board work?

M1kerochip, 2019-11-03:
This is the board: (It's maked as a backpill, which you specificially mention)

https://robotdyn.com/diaphragm-mini-pump-for-liquid-12v.html
and pinout:
https://robotdyn.com/pub/media/0G-00005692==STM32F103C8T6-STM32MiniSystem/DOCS/PINOUT==0G-00005692==STM32F103C8T6-STM32MiniSystem.jpg

I found a generic bluepill pinout:
https://idyl.io/wp-content/uploads/2018/07/stm32f103-pinout-diagram.png

And it seems to be the same.

I quite like the build quality of these manufacturers. They seem to be a step above the usual Chinese made dirt cheap boards.

keirf, 2019-11-03:
That board looks good. My problem has been with a cheaper "Black Pill" which has the STM32 at 45-degree rotation same as Blue Pill. The one you link looks better quality and the STM32 chip as in the product photos looks genuine.

keirf, 2019-11-03:
I will update my suggestions based on work I've done on identifying STM32 fakes in the last few days. There are fake marked STM32F103C8T6 devices out there, but the ones I've see are physically distinguishable and I also now have a test firmware which can spot them.

M1kerochip, 2019-11-03:
Good stuff.

I’ll order one of those boards too so. (I have a cheaper €2 Chinese blue pill board on the way).

Yes, they do seem to charge more than the usual Chinese knock-offs, but everything looks good/genuine. I’ve only bought about 10 or so of their boards, but never had a single problem.

Volutar, 2019-11-05:
Isn’t it better to add “blinky” testing into the main firmware? It could make all these tests (not infinite loop for sram, maybe just couple), and store results for later report. And when software is run - it could report about hw issues. Also, command line parameter with all necessary info about firmware and test results (even if it’s ok) also would be nice. Though reflashing rom each time is quite bad idea. Well, then command line parameter “testhw” would be quite handy. Or “testflash” specifically for 64k capability (and leaving other tests to default firmware startup sequence). 

I have the blue pill with small text on the chip (which is stated as probably not genuine) but blinky is blinking. So probably it works as expected, but i’m not really sure (since there I didn’t see any reports, bad or good ones).

keirf, 2019-11-05:
That would be nice but my time is finite. If you want to see your test report, log into serial via same serial adapter you used for programming Blinky. For example, using PuTTY, or whatever client you prefer. At 115200,8n1. Now reset the Pill board and you should see your report printed out.

keirf, 2019-11-05:
Also put a photo of your STM32 chip here, for interest.

Volutar, 2019-11-05:
[image]
Okay. Here you go. I had hard time of trying to make it readable, since it's "black on black".

And i used stlink device to program, so there's no com port for puttying :(

keirf, 2019-11-06:
Yes that looks like a genuine chip anyway -- there is only the single circular mark, rather than three.

closed 2019-11-06 by keirf

Volutar, 2019-11-06:
Single depression, yes. But text mark is smaller and hardly readable. So I guess text size shouldn’t worry as long as there is only single 1 pin circular mark. Perhaps wiki requires smallest addition.

keirf, 2019-11-06:
I will add that the only way to be sure is to run Blinky.

Universal-Solder, 2021-03-07:
Hi, I don't know how to use your "blinky". Isn't there a .bin file that I can flash using ST-Link Tool, for example? Or an Arduino code? 
thanks

keirf, 2021-03-08:
The Blinky HEX is in the Greaseweazle release zip file under the alt/ subfolder.

Universal-Solder, 2021-03-13:
Thank you so much! Blinky is a great help. Now I am even more confused than before since most of the chips I expected to be fakes seem to be genuine.

Universal-Solder, 2021-03-13:
I have 2 different chips with marking 991KA 93 MYS 803. One of them seems to be genuine, the other one is fake. 
Oh my, this is bad.

keirf, 2021-03-13:
A lot of fakes work well enough to pass Blinky. That's one reason I wrote it, as all you'll get on Chinese blue pill boards is fakes. So a method is needed to sort good fakes from bad. If they pass, they are usable in my projects.