keirf/greaseweazle #3: How does it work?

davidgiven, 2019-11-03:
Hello --- I have a similar project based around a PSoC5 board: http://cowlark.com/fluxengine

I'm very much wanting to ditch the expensive and very closed source PSoC5 development board in favour of different hardware; I'd looked at the Blue Pill and didn't think it was powerful enough. So I'm really surprised to see that you're doing exactly what I wanted!

How does the Greaseweazle (nice name, BTW) work? I see you're using the DMA engine to accumulate samples and then the CPU is sending them via USB --- but where's the DMA engine getting them from? My concern with the STM32 is that I didn't see any hardware that would time pulses for me, but the CPU wasn't powerful enough to service an interrupt for every pulse. Have you found a way to repurpose some hardware component? If so, what are you doing for sample playback for write support?

And isn't the STM32 a 3.3V device? Are you having any problems there?

keirf, 2019-11-03:
STM32 devices have a set of quite powerful general-purpose timers. These can be configured to "input capture" (take a counter timestamp on an input pin edge) or "output compare" (toggle/set/clear an output pin based on comparison between counter and a threshold value). DMA can be triggered on various timer events, and DMA is highly programmable so you can set up buffer rings of flux values in/out, and have occasional IRQ events to batch-service those rings.

The chip also has a good number of 5v tolerant pins. So I use those on the floppy bus, 1k pullups on inputs (or even using the weak internal pullups seems to work okay), and open drain outputs.

davidgiven, 2019-11-03:
Re timers: yes, that's ideal --- that's essentially what I built in rather dodgy Verilog. What about output support? Are you abusing a PWM system?

Re 5V tolerant pins: I'm mainly worried about sending signals to the drive. The datasheets I've seen suggest high logic thresholds of about 2V or 2.5V, but the floppy bus signalling is so underspecified that I've no idea if there are drives with a higher threshold. I've had one report of a strange issue where a user wasn't getting any response at all from a drive unless both an A and a B drive were installed --- termination issues, maybe? This isn't an area I have a lot of experience with, and the whole area of floppy drive electrical interfaceis dubious enough I've been focusing on 5V only.

If you're getting good results with 3.3V, then that's good news, because I've been having serious trouble finding alternative 5V hardware.

What value are the internal STM32 pullups?

keirf, 2019-11-04:
Yes I use PWM for output.

The floppy bus is open collector and drives have pullups on their inputs. So GW can simply configure its outputs open drain. Because I use 5v tolerant pins, the drive pullups can pull up all the way to 5v.

The stm32 pullups are 30k-50k. Amazingly they work even for the data line, but would be better using external stronger pullups. Also it is inadvisable to use the internal pullups as the 5v tolerance is affected.

davidgiven, 2019-11-04:
Clearly I need to read up on this more. An output line which isn't being driven will float up at both ends, so I'd expect there to be a current flow from the drive (5V) to the STM32 (3.3V). Won't this be flowing through both pullup resistors backwards? I really don't know what the implications of this are, esp. if there's anything like a diode in the circuit.

Your instructions on assembling the Greaseweazle don't mention external pullup resistors...

How stable is your USB protocol? Because I'd quite like to assemble one of these and then add support to the FluxEngine client for sampling from a Greaseweazle directly. It'd make a really good alternative to the PSoC5 hardware (and let me do comparative sampling --- I'm having trouble getting good reads from Mac CLV disks and am wondering whether it's a drive limitation or a bug in my sampling code).

drdpj, 2019-11-04:
The drive end is pulling the outputs up to 5V, the STM32 is configured to
be open drain on the outputs, so you're either running high
impedance (pulled up to 5V by the floppy end) or drain (down to 0V through
the STM32) - nothing should be floating?

FWIW, I'm running this on both 3.5" and 5.25" HD floppy drives and it's
behaving fine.  Keir, am I reading the code right, is it sampling at 72MHz?

davidgiven, 2019-11-04:
Sorry, 'floating' was the wrong word there. I mean, being pulled up by the resistor to the voltage rail.

drdpj, 2019-11-04:
Gotcha. You never see 3.3V at the STM32 end on open drain outputs, it's
just very high impedance at that end and can happily cope with the 5V being
wafted at it.
https://www.st.com/content/ccc/resource/technical/document/reference_manual/59/b9/ba/7f/11/af/43/d5/CD00171190.pdf/files/CD00171190.pdf/jcr:content/translations/en.CD00171190.pdf
-
pg 163/164.

keirf, 2019-11-04:
Outputs from STM32 to the drive I think are covered above.

Inputs from the drive to the STM32, best approach is to make those pins "floating input" and have an ext 1k resistor pulling up to 5v. The simple approach is to make those pins "pull-up input" but you do then get current leaking to Vdd (3.3v). I measure 4.5v at such pins, meaning 0.5mA leakage. Actually although out of spec it works okay -- most Gotek inputs aren't 5v tolerant in the first place and they work reliably enough.
Either way, Greaseweazle detects external pullups during initialisation, and configures "floating inputs" if it can.

Greaseweazle does indeed sample at 72MHz (14ns) resolution. The USB protocol is good for 6-7Mbps as long as there's not other Full Speed devices on the same hub hogging USB slots. My tests show I can write flux intervals as low as 1.25us somewhat reliably (ie. with some automatic detect-and-retry). This is more than fine for even HD disks.

closed 2019-11-05 by keirf