keirf/greaseweazle #4: Would this work with my STM32F103C8T6-based custom board?

roybaer, 2019-11-04:
Hi!

I would like to ask whether there is any chance that your firmware might work with [my custom board design](https://github.com/roybaer/usb_floppy_adapter) from early 2018.

It uses the same STM32F103C8T6 and plugs into the back of the drive. Unfortunately, I never found the time to write proper firmware for it.

The pinout is:

µC pin | Signal | Drive pin
----|----|----
PA0 | INDEX | 8
PA3 | WGATE | 24
PA4 | DRVSA | 14
PA5 | DRVSB | 12
PA6 | MOTEB | 16
PA7 | MOTEA | 10
PA8 | DSKCHG | 34
PA9 | REDWC | 2
PB0 | DIR | 18
PB1 | STEP | 20
PB10 | TRK00 | 26
PB11 | WPT | 28
PB12 | DTID1 | 4
PB13 | SIDE1 | 32
PB14 | RDATA | 30
PB15 | WDATA | 22

keirf, 2019-11-04:
No reason why not really. Most of the pin-specific stuff is at the top of src/floppy.c. Gotchas I can see:

- Not all your pins are on GPIOA so you will need to be more flexible than my single 'gpio_floppy' defn.
- INDEX gets configured for an EXTI IRQ. Yours is on PA0 so you need to adjust the EXTI init appropriately, and your IRQ number will be different.
- The timers and DMA channels for RDATA/WDATA will be different. That's mostly abstracted at the top of src/floppy.c but the code currently assume two separate timers, whereas you are on two channels of the same timer. It shouldn't much matter, but you'll have to juggle the init/deinit code to do a fuller initialisation when switching between read and write mode.

roybaer, 2019-11-04:
Nice!  That should be doable.  I'll have a look in the near future.

The use of two different GPIO ports was mostly for routing reasons.  The board is only 19mm × 48mm, after all.

closed 2019-11-05 by keirf

Volutar, 2019-11-05:
Sorry for making comment in this (closed) issue, but I think this question doesn't deserve its own, and practically related to this.

So question is:
Is it possible to rewire blue pill differently (according microcontroller pin reassigning), to minimize "dupont wiring", so it will be possible to adjust STM32 firmware?
To something like this:
[image]

Currently it looks randomish... And making GW "that" will definately ease wiring alot.

drdpj, 2019-11-05:
You need to see which pins are 5V tolerant... Not all of them are!

Volutar, 2019-11-05:
> You need to see which pins are 5V tolerant... Not all of them are!
There is only one pin - B5, which is not tolerant on this side of bluepill, and it's "not connected".

keirf, 2019-11-06:
A11 and A12 are USB. A9 and A10 are for serial (ie at least I use them!). It is not tenable long term to expect users to connect dirdct to the pill. Adapter boards will happen. They will be very cheap.

Volutar, 2019-11-06:
Well, it’s a bummer. While A9/A10 can get some workaround, A11/A12 are not usable :(
People with dozen of disks to capture hardly gonna order adapter boards. I’m sure >90% of users will use mishmashed dupont cables (and already using it). So at least some wire rearrangement ( in order to lessen crossing) would be appreciated. 
Thanks for the answer!

keirf, 2019-11-06:
Those people will just have to put up with a slight Dupont mishmash to save a couple of quid. It's not actually that bad anyway.

dwillmore, 2020-04-27:
Could A8/A9 be used for the other drive select and motor signals?