keirf/greaseweazle #5: Empty SCP....

drdpj, 2019-11-06:
I've got a 5.25" floppy that's producing an empty SCP file (others are working)...  Did a kryoflux dump just to make sure that it's actually got something on it and I haven't wiped it by mistake! I've uploaded it here (bcpl-sag-kryo and scp):
https://drive.google.com/open?id=1DghUX-RZHiZ5gKnVT8VmsotsdAvuJNUI
I've stuck 1k pullups on the inputs now, if that's relevant :)

keirf, 2019-11-06:
Is this taken since adding the pullups?

drdpj, 2019-11-06:
yup

drdpj, 2019-11-06:
-> it seems (having now got the lid off), that it's not seeking to track 0 before it starts reading, so it just sat at the last track it read (which was 81 on a previous disc)...  I'll try it a couple more times..

drdpj, 2019-11-06:
Yup, not seeking back to zero with the greaseweazle, but it is with the kryoflux...

Volutar, 2019-11-06:
That might be related to delay_params.step_delay being set to 3ms, which works for 3.5", when 5.25" drives require 6ms minimum (some older models work only with 12ms+).
There's commented line 401: #set_delays(step_delay=3) in gw.py. Perhaps uncommenting and changing value to 6 will help.

keirf, 2019-11-06:
Did you make yourself an adapter board? I can't remember. Anyway it sounds like greaseweazle trk0 pin is stuck low. So it thinks its already on track 0. Wiring?

keirf, 2019-11-06:
Or a bad pull up connection on that line?

drdpj, 2019-11-06:
Bad pullup - how terribly embarrassing! the thing had gotten doinked :D (I've bodged the resistors onto the bluepill, I'm running adaptor free :))  As you were :D

closed 2019-11-06 by drdpj

keirf, 2019-11-06:
We need the proper adapter boards. Can't be much longer

Volutar, 2019-11-06:
Wait. What? I have no issues with bluepill alone, without any extra pullups. Are there some special conditions requirig for extra circuitry? Probably redraw a scheme of HW connections?

keirf, 2019-11-06:
@Volutar The internal STM32 pullups aren't really strong enough, properly speaking a 1k resistor pullup on lines into the STM32 are recommended. Certainly that's what I've put on my adapter boards.

But actually in practice the STM32 pullups have been working well enough for everyone using naked Pill boards so far :)

If you do want to install pull ups, you must do it on *all* of the following lines: B3,B6,B7,B8. Greaseweazle detects the external pullup on B6 at startup and disables the internal pullups on all four pins.

Volutar, 2019-11-06:
@keirf  I've programmed Arduino for floppy dump (recently, it's still WIP), and only RDATA line did require for pullup, every other line didn't require that. On both 3.5" and 5.25" drives I've used. I tested that with oscilloscope, and there was pretty obvious that 1K is necessary. And looking into different FDC schemes - only RDATA was with 1K pullup.

So if I get this issue right, Greaseweazle detected that there is the external pullup at B6, and simply disabled for all of them, including TRK0, so @drdpj ran into this issue, because not all of them were properly attached. Right?

keirf, 2019-11-07:
That is correct.

drdpj, 2019-11-07:
And one blob of solder later, it's back in the room :)