keirf/greaseweazle #7: Index stability

Volutar, 2019-11-09:
When using 5.25" drive, some revolutions instead of 14mln ticks take 262k ticks only. It's exactly the 3.6ms - /INDEX pulse width. I have a theory why it happens.
Being making disk reading hw/sw myself, I've noticed that sometimes index impulse, when returning back from 0 to 1, sometimes make sporadic jumps back to 0. Like it's prolongated under capacitor with some noise influence. So personally I made "timeout" value to avoid these "too short rotations" (by skipping first 10ms after /INDEX is received). I hope you will fix that.

And by the way, I found that 5.25" are making no stepping actions until I attach analyzer to /DIR or /STEP pin. Default voltage is 5.25v, and pulsing down to 0.3v when measured (and I don't know how to measure original voltages without analyzer resistance).

keirf, 2019-11-10:
Have you attached a scope or analyser and observed what these glitches look like and how long they are. There are a few options for how to filter them out. One is to use the filter hardware on the timer channel at the INDEX input pin. But that's only feasible if the glitches are very short. But yes we can filter these out one way or another.

Regarding /DIR and /STEP, my guess would be that the pull-up resistors at the drive inputs are too strong for STM32 to drive against. These old drives may well have a discrete resistor pack and you may be able to measure resistance of say /DIR to 5v with the drive powered off and disconnected.

The fix if this is the problem is going to be external buffering. You will need an open-drain/collector transistor buffer, or an open collector driver (eg 74LS38, though there may be a better choice as we don't need NANDs) on every output from STM32 to drive. If this inverts the STM32 outputs I'm happy to add some way to make Greaseweazle invert its outputs.

keirf, 2019-11-10:
By the by, have you considered finding a relatively more modern 5.25 drive to use? Bouncy index sensor and strong pull ups probably indicates an older drive and later drives can be better for dumping anyway, as well as being easier to interface?

Volutar, 2019-11-10:
1. I already described how it looks like. It happens due to optical sensor schematics. It opens instantly, and closes with some kind of digital “diffusion” when watching on logical analyzer graph. Something like 11111100000000000000001001101111111111. The length of that transition period is short. And all what is needed - is to simply disable this pin processing for like 10ms (just in case) after index been detected. And transition period actually doesn’t matter with that filtering strategy (I used it with arduino). 

2. Extra schematics and ICs for for STEP to work? Well, its not an option for me. It even works when I hookup ONE wire of floating (not even powered) logic analyzer! It’s way easier than making all these constructions. Extra schematics is not what I expected, considering the fact that arduino, which I’m programming exactly for same purposes (making universal floppy dump), works with just single 1K resistor to pullup RDATA. And having no extra pullup resistor on STM32 made me happy. And reading “need of buffer IC and transistor” stuff is bit disappointing to say the least :(

3. Finding another 5.25”, and then another, because _noone_ will guarantee that it will work with this microcontroller “limitations” - is the worst way possible. I’ve had hard time of getting even this one. So I’ll have to deal with what I have (which actually works perfectly inside real hardware and with arduino).

drdpj, 2019-11-10:
For reference, it's worth noting what Chuck did to ensure it works with
older drives here on the VCFED forum:
http://www.vcfed.org/forum/showthread.php?72154-Open-source-flux-reader-writer-running-on-2-bluepill-board&p=593189#post593189

(i.e. buffered it :))

Volutar, 2019-11-10:
So will index filtering happen?

Volutar, 2019-12-17:
Really, it’s not that hard to add noise cancelling algorithm for optic sensor bounce. Considering the length of the “hole” is known and period of signal is also not 0...infinity.

Volutar, 2020-08-20:
Resolved with https://github.com/keirf/Greaseweazle/commit/ff635cf5b79e8326fa8745792ffca074fe7be47b

keirf, 2024-09-23:
@Volutar I don't know if this one is fixed, since your problem was bounce during deassertion of INDEX. Whereas #52 was a bounce during assertion of INDEX, and that's what I fixed, with a 50us "mask".

The problem is that the firmware doesn't know the expected period of INDEX. It can be shorter than the pulse width on a regular soft-sector drive and disk, if for example we are sampling all holes in a 32-hard-sector disk (then the shortest INDEX-to-INDEX time can be 167ms/32/2 ~= 2.5ms).

However, I can do better than the current algorithm. So the question is -- did you ever confirm that this issue is fixed on your hardware?

closed 2024-09-24 by keirf