keirf/greaseweazle #10: The name Greaseweazle name

staticmem, 2019-11-28:
I have to ask how you came up with the name for this project,  I've googled it but no obvious answers,  sorry if it's been mentioned before but I've seen nothing that explains it.

keirf, 2019-11-28:
I always used to want a Catweasel floppy controller, and so this is partly a tribute to that. A greased weasel is an amusing term and has a nice assonance, and Catweazle was a dodgy 70s UK TV series. So I sort of put that all together to get the weird memorable name!

staticmem, 2019-11-29:
Ah yes, Catweasel.  I remember watching the series when I was a young lad in the late 60s when living in NL at the time.  We moved there for my father's work for a year or so.  Bud Tingwell (the father in series) was an AU actor probably better known later for the movie The Castle. :)

closed 2019-12-23 by keirf