keirf/greaseweazle #11: Beware of 4 MHz STM32 models

nretro, 2019-11-30:
I bought a STM32F103C8T6 from China on ebay. On close examination, the description says 

4-16MHz crystal.

and in fact has a 4 MHz instead of an 8 MHz crystal. The blinky test succeeds nicely, but only at half speed (output at 57600 baud).

[image]

keirf, 2019-11-30:
I figure I ought to be able to measure HSE against the HSI or LSI oscillators and then set things up correctly for 4 vs 8 mhz crystal...

turnkit, 2019-12-25:
Is there a proper forum for non bug issues?

Will the black pill work just fine too?

https://www.cnx-software.com/2019/12/24/stm32-black-pill-board-features-stm32f4-cortex-m4-mcu-optional-spi-flash/

keirf, 2019-12-26:
There is a Facebook group. The black pill should work but note the pin arrangement is different. The few black pills I purchased had fake stm32 which didn't work.

cbrunschen, 2019-12-26:
I think @turnkit meant the different, new "black pill" that is based on an STM32F4x1 chip, see [the link they posted](https://www.cnx-software.com/2019/12/24/stm32-black-pill-board-features-stm32f4-cortex-m4-mcu-optional-spi-flash/), with a USB-C connector for visual differentiation if nothing else.

keirf, 2019-12-26:
It would need porting but could then work. I didn't spot any of these new ones on ebay yet.

cbrunschen, 2019-12-26:
It seems they're mainly available on Aliexpress (for example [1](https://www.aliexpress.com/item/4000084631884.html), [2](https://www.aliexpress.com/item/4000103610226.html), [3](https://www.aliexpress.com/item/4000138305460.html), [4](https://www.aliexpress.com/item/4000068998794.html)) at the moment.

ptor, 2020-01-08:
If someone who has ordered a verified working one (a blue, or the new black pill) from a "cheap" source (ebay or Aliexpress) could link to where they bought it from, I would be eternally grateful. I've been on the fence for a long time since I simply don't dare buy one - it looks like it's hit or miss.

nretro, 2020-01-08:
Just get a Greaseweazle F730 DYI Kit from Keir. He posts them for a couple of pounds. Just send him an email. Not the cheapest way, but you definitely get a working one while supporting the project.

I've also made some progress on the 4MHz problem. Due to PLL limitations, you only can get the max speed of 72MHz on the STM32, when using 6, 8 or 12 MHz crystals. But it's easy to source these crystals cheaply and replacing them should not be a problem.

keirf, 2020-01-17:
Not much more to say here then. I haven't had any more reports of 4MHz crystals. If there are then I will consider updating the docs, or make Blinky aware of it.

closed 2020-01-17 by keirf