keirf/greaseweazle #13: Python Syntax error

staticmem, 2019-12-23:
I'm running an upto date LinuxMint system.  I'm having issues running the following:

staticmem@asrock:~/Builds_64/Greaseweazle-0.7/scripts
$ python gw.py read --ecyl=39 --single-sided mydisk.scp /dev/ttyACM0
  File "gw.py", line 20
    print("Actions: ", end="")
                          ^
SyntaxError: invalid syntax

I don't have pip3 on this computer but have: pip, pip2 and pip2.7.

sudo apt install python3
python3 is already the newest version (3.6.7-1~18.04)

sudo pip3 install crcmod pyserial
env: ‘pip3’: No such file or directory

From what I read pip3 is only provided in customised versions of python3?

Is this pip3 issue the problem or something else?

staticmem, 2019-12-23:
The answer in the link solved the pip3 problem:
https://superuser.com/questions/769920/python3-pip-installed-but-pip3-command-not-found

But still having issues:
sudo pip3 install crcmod pyserial
The directory '/home/staticmem/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/staticmem/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Requirement already satisfied: crcmod in /usr/local/lib/python3.6/dist-packages
Requirement already satisfied: pyserial in /usr/local/lib/python3.6/dist-packages

I have no idea what the actual problem is.

keirf, 2019-12-23:
The original syntax error is because you need to run python3: eg `python3 gw.py read ....`

The pip3 error is something to do with file/folder ownership checks and it doesn't like root (superuser) trying to write into your home directory. Try `pip3 install crcmod pyserial` or `pip3 install --user crcmod pyserial`

staticmem, 2019-12-23:
I tried both those pip3 install lines and but appear to fail...
 
pip3 install --user crcmod pyserial
Collecting crcmod
  Using cached https://files.pythonhosted.org/packages/6b/b0/e595ce2a2527e169c3bcd6c33d2473c1918e0b7f6826a043ca1245dd4e5b/crcmod-1.7.tar.gz
Collecting pyserial
  Using cached https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl
Building wheels for collected packages: crcmod
  Running setup.py bdist_wheel for crcmod ... error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-imoji3re/crcmod/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpuqr2hmrepip-wheel- --python-tag cp36:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
 
  error: invalid command 'bdist_wheel'
 
  ----------------------------------------
  Failed building wheel for crcmod
  Running setup.py clean for crcmod
Failed to build crcmod
Installing collected packages: crcmod, pyserial
  Running setup.py install for crcmod ... done
Successfully installed crcmod-1.7 pyserial-3.4

I also tried running the GW script using python3 instead of python.

python3 gw.py read --ecyl=39 --single-sided mydisk.scp /dev/ttyACM0
Traceback (most recent call last):
  File "gw.py", line 24, in <module>
    mod = importlib.import_module('greaseweazle.tools.' + argv[1])
  File "/usr/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/staticmem/Builds_64/Greaseweazle-0.7/scripts/greaseweazle/tools/read.py", line 12, in <module>
    from greaseweazle.tools import util
  File "/home/staticmem/Builds_64/Greaseweazle-0.7/scripts/greaseweazle/tools/util.py", line 12, in <module>
    from greaseweazle import version
ImportError: cannot import name 'version'

$ ls -l ~/.cache/pip/
total 8
drwx------ 7 staticmem staticmem 4096 Dec 23 18:20 http
drwxrwxr-x 3 staticmem staticmem 4096 Dec 23 18:20 wheels

Do I need a "wheels" group?  I don't see it in /etc/group.

keirf, 2019-12-23:
This new pip3 install error is not one I've seen before. A quick Google suggests you need `pip3 install wheel` or perhaps `pip3 install --user wheel`.

The python3 error is apparently a missing file `version.py`. It is distributed with Greaseweazle but I did forget to include it in at least one release! So re-download latest Greaseweazle and try again.

staticmem, 2019-12-23:
I tried those other pip3 install commands and they report error messages.

The version.py is in the 0.7 release already.
I think my Python installation has become a complete mess.  I recall having issues with it once before.  Asking apt to remove it (so that I could re-installl from fresh) it gave me a severe warning that this may break the system.

I've tried deleting files and reinstalling various packages including the home account area but nothing I do makes it work.  I'm guessing there is some version conflict problem happening.

staticmem, 2019-12-23:
This seems to work as long as not run with "sudo":

pip3 install crcmod pyserial
Collecting crcmod
Collecting pyserial
  Using cached https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl
Installing collected packages: crcmod, pyserial
Successfully installed crcmod-1.7 pyserial-3.4

staticmem, 2019-12-23:
I read in various sites never to use "sudo" with "pip"

https://itsfoss.com/python-setup-linux/
See: Big Caveat

So as a last ditch effort I wondered if I could reverse the damage so I ran these as "sudo" to uninstall

sudo pip3 uninstall crcmod
sudo pip3 uninstall pyserial

I had to repeat the commands a few times as it deleted files in root area and the home account.  I kept to doing it until it could not find any other installed files.

Then I ran the pip3 install line again but as a normal user.

pip3 install crcmod pyserial

staticmem@asrock:~/Builds_64/Greaseweazle-0.7/Greaseweazle-v0.7/scripts
$ python3 ./gw.py read mydisk.scp /dev/ttyACM0
** Greaseweazle v0.7 [F1], Host Tools v0.7
Reading Track 0.0...Command Failed: Track 0 not found

Bluepill not connected not connected to a drive yet but at least that looks promising now.

@Keirf:  I know very little about Python but maybe you need to remove the "sudo" from the "pip3" install commands?

keirf, 2019-12-23:
Ok that looks good! I think you're right, and actually non-root `pip install` automatically gets converted to `pip install --user` which is safe and won't mess with your system-wide Python installation. I will look into converting the recommended command lines.

keirf, 2019-12-23:
Fixed. Thanks for the feedback.

closed 2019-12-23 by keirf