keirf/greaseweazle #15: idVendor and idProduct USB attributes

staticmem, 2020-01-20:
I want to assign a static value for the Greaseweazle under Linux.  When I run these commands I see the following values. Which one of the Vendor and Products is the actual GW application and what is the other one for?

$ udevadm info -a -p  $(udevadm info -q path -n /dev/ttyACM0) | grep -i idvendor
    ATTRS{idVendor}=="1209"
    ATTRS{idVendor}=="1d6b"

$ udevadm info -a -p  $(udevadm info -q path -n /dev/ttyACM0) | grep -i idproduct
    ATTRS{idProduct}=="0001"
    ATTRS{idProduct}=="0002"

keirf, 2020-01-20:
1209:0001 is the Greaseweazle board -- 1d6b:0002 is a USB 2.0 hub which must be on the path to Greaseweazle (could be part of your chipset if not an external physical hub).

But there are better ways to match on Greaseweazle, specifically the product string (and optionally manufacturer string): see scripts/49-greaseweazle.rules for an example usage in a udev script.

Regarding python vs python3, the former seems to be correct on Windows. You are correct it should be python3 on Mac and Linux. I should probably have a page on "Invoking GW" and then make it clear I use Windows command lines from then on.

staticmem, 2020-01-20:
I already copied those rules to /etc/udev/rules.d/.  That works fine, what I wanted was a way to assign the device to a known STATIC device name.  The problem is I may have different ACM devices plugged in.  Depending on the order of being plugged in the assignments will vary and assigned as /dev/ttyACM0 or /dev/ttyACM1 or some other device so then need to go hunting for it.

I'll try this method as shown at answer 43:

https://unix.stackexchange.com/questions/66901/how-to-bind-usb-device-under-a-static-name

staticmem, 2020-01-21:
Based on the link above I used this in place of original rule:

ACTION=="add", ATTRS{idVendor}=="1209", ATTRS{idProduct}=="0001",SYMLINK+="greaseweazle"

Then run:
sudo udevadm control --reload-rules && udevadm trigger

Unplug and Replug the device.

Device is now "/dev/greaseweazle"

This may not be a perfect solution but works for one GW instance, if others are connected it probably won't.

keirf, 2020-01-21:
Oh that's a good idea. Of course the action (SYMLINK) can be coupled with the smarter matching (manufacturer, product). Arguably I should add a serial number, then you could disambiguate multiple instances on that.

staticmem, 2020-01-21:
Yes, the Serial number would be a good idea to allow different units plugged to all get their own unique device assigned regardless of the USB device used.

staticmem, 2020-01-21:
The STM micros each have a unique serial number that I'm guessing could be used for the serial number.

keirf, 2020-01-21:
Yes I can cook it down into base64 or something.

keirf, 2020-01-21:
This is all fixed in v0.11

closed 2020-01-21 by keirf

staticmem, 2020-01-22:
I'm confused as I can't see any sort of serial identification in the new udev rules.  the symbolic link of /dev/greaseweazle can only point to 1 Greaseweazel instance.  If I had two GWs connected at the same time what would the other symbolic link be?

keirf, 2020-01-22:
There isn't any. Obviously if you want pretty static names for your specific weazles, you're going to need to create custom rules which refer to your specific serials.

Or it would need more machinery (eg udev triggered python script) to dynamically create new pretty symlinks for each new serial seen.

Now gw.py has auto probe, there's also scope for identifying serials by pretty names purely from gw.py eg with a config script there. This would be more generally useful (eg Windows)

stuaxo, 2021-08-31:
Symlinks like /dev/greaseweazle0  /dev/greaseweazle1  would be nice for sure.