keirf/greaseweazle #16: 2DD 5.25 360k ibm-pc format

brainy78, 2020-01-25:
Hello!  Sorry for my English... I'm Korean.

First of all, Thank you so much!

So many Korean people whose hobbies are collecting retro IBM-PC are glad to meet your project!
(I'm vice manager of online-community for "DOS/win3.1")

Your core theme of the project is Shugart, but IBM-PC format doesn't seem to be out of your interest... T.T

I know 5.25 shugart pin-out is different from that of 5.25 IBM.

So that?? 2DD 5.25 360KB  ibm-pc formatted diskette does not working using greaseweazle. It means reading and writing does not operate well... 

Error Message shown as bellow,
"Greaseweazle v0.11 [F1], Host Tools v0.11
Reading Track 0.0...Command Failed: Seek: Track 0 not found"

360KB 2DD formatted diskettes **don't work with both 5.25 2HD and 2D drives.**
(But 2HD 5.25 1.2MB IBM-PC formatted diskettes **do work well** with 5.25 1.2MB 2HD floppy drive.)

So I have question for you, weather 360KB 2D IBM-PC format is not your intetest yet or I did something wrong...

If 5.25 2D 360KB IBM-PC format works good with greaseweazle, So many Korean people will be very grateful to you.

keirf, 2020-01-26:
Are you directly connecting a Blue Pill to your drive with Dupont wires (not using an adapter board)? Then  check your connections carefully. The following pins must be correctly connected for Track 0 initialisation to work properly: Pins 12 (SELECT), 18 (DIR), 20 (STEP), 26 (TRK0).

The above is most likely problem (simple bad connection) if you are directly connecting. Easy to makea mistake in the mess of wires.

Some other possible problems (in descending order of likelihood):
1. Drive select is not on pin 12 (DS1) on your drive, but on pin 10 (DS0).
2. Input termination on your drive is 150 ohm instead of 1kohm. The MCU cannot drive directly into 150 ohm, you would need to add buffering, or remove that termination pack (possibly replacing it with 1kohm pullups).
3. Possibly you need input pullups of 1k to +5v on pins 8 (INDEX), 26 (TRK0), 28 (WRPROT), 30 (RDATA).

Other things to try are a more modern 5.25 drive. The old media will read/write fine. The incompatibilities generally come from the software side, but Greaseweazle doesn't mind for example 360rpm vs 300rpm spin rate, and when dumping disks you should still get a good analysis in software such as HxC track analyser (https://hxc2001.com/download/floppy_drive_emulator/HxCFloppyEmulator_soft_beta.zip)

brainy78, 2020-01-26:
I bought the drive adapter and connected to it. So I'm sure that hardware assembly is OK, because 1.2MB HD Drive works perfectly(to read and then write back works well).

But with all same condition, Exchanging into 360K 2D Drive does not works with message "Track 0 not found".

I tested three good 2DD FDD with GW, but all of them do not works properly.

As you mentioned earlier, to read 360K floppy diskette with 2HD 5.25 FDD will be OK?

and then Is there optional parameter of gw.py?

I read 360K floppy diskette with 2HD 5.25 FDD, and then I wrote soon on diskette with that image.
But that diskette didn't recognize properly in FDD of any PCs. 

I just could see the directory list but any files didn't be executed and copied showing "reading data error". (but 1.2mb 2HD FDD format is OK! to read and write back)

I know that 1.2mb format has 80-cylnder but 360kb format has 40-cylnder. 

So to read 360Kb fIoppy diskette, I used --ecyl=39 option, but FDD header stopped in the middle of surface. Of course that Image could not be read well... 

Is there something missed I have?

bradwh, 2020-01-27:
There is complication using 360Kb disk with the 1.2Mb 80 Track drives.
The track width on the 1.2Mb drive is half the width of the 360Kb 40 track drive. So writing a 360Kb  disk with the 1.2Mb drive produces a half width track compared with the 360Kb 40 track drive.
This causes compatibility issues when writing, reading is not a problem. The most reliable method of writing 360Kb disks is to use a 360Kb 40 track drive, this is regardless of the method of controlling the drive. 
Hope this helps.

drdpj, 2020-01-27:
I have used both the F1 and  F7 versions o read disks with a 1.2mb drive
and write them with a 360k 40 track drive. It works fine. Make sure you
have the right drive numbers set on the drives. Depending on the drive you
might need to check how the head load and motor jumpers etc are set too.
Without seeing your setup it's hard to say what is wrong exactly, but it's
not the Greaseweazle.

brainy78, 2020-01-27:
> There is complication using 360Kb disk with the 1.2Mb 80 Track drives.
> The track width on the 1.2Mb drive is half the width of the 360Kb 40 track drive. So writing a 360Kb disk with the 1.2Mb drive [...]

Thank you very much. 

You give me a logical reasoning. 

So, All that's left now is to recognize 360K FDD with greaseweale!

^^

> I have used both the F1 and F7 versions o read disks with a 1.2mb drive and write them with a 360k 40 track drive. It works fine. Make sure you have the right drive numbers set on the drives. [...]

Thank you so much! This is what I want to see. I did not find any information relevant 360K. You confirm me. I worried about not supporting 360K! Thank you. I try to recognize 360K FDD.

Thank you, Thank you everyone!!

bradwh, 2020-01-27:
You should not have any problem using Greaseweazle to read 360Kb disks on either a 360Kb 40 track drive or a 1.2Mb 80 track drive. I have done both with Greaseweazle with no problems.
However if you want to write 360Kb disks to be read in any other drive, you must use a 360Kb 40 track drive. Using an 80 track drive will produce poor results. This is not a Greaseweazle problem, it is a case of track width problem and a known problem with this old format. You need to source a 40 track drive to sort this out.

brainy78, 2020-01-27:
> You should not have any problem using Greaseweazle to read 360Kb disks on either a 360Kb 40 track drive or a 1.2Mb 80 track drive. I have done both with Greaseweazle with no problems.
> However if you [...]

Thank you for your concerning!! ^0^

Bellow is good to me!
The 2HD 5.25 1.2MB IBM-PC formatted diskettes do work(read and write) well with 5.25 1.2MB 2HD floppy drive.

But bellow is bad to me!
1. 360KB 2DD formatted diskette images didn't write properly with 5.25 1.2MB 2HD FDD.
(Maybe 
2. (Bigger problem) 360KB FDD didn't read & write with GW.
"Greaseweazle v0.11 [F1], Host Tools v0.11
Reading Track 0.0...Command Failed: Seek: Track 0 not found"

I failed to operate four 360K FDDs, which I have (brand: two goldstar and two HP)
I tested on win7 & win10 with GW version 0.11. 
The jumpers DS1, MM is closed.

Oddly, as I just exchange 360K to 1.2MB with exactly same condition, both to read & wtite is fine.
But conneted with 360K FDD allways shows message bellow,
"Greaseweazle v0.11 [F1], Host Tools v0.11
Reading Track 0.0...Command Failed: Seek: Track 0 not found"

What the matter with me? T.T

Anyway, I'll keep doing my best to success to operate 360K with GW!

Thank you everyone! ^^ Thank you Keirf for your FF & GW!

bradwh, 2020-01-27:
Ok, Lets establish the picture.
Forgetting the 1.2Mb drive for the moment.
You have a single 360Kb drive configured as DS1 connected to the plug after twist in the cable.
Now in your command line you issue the command 
"python gw.py read --drive=a --ecyl=39 mydisk1.scp"
then you get the error "Reading Track 0.0...Command Failed: Seek: Track 0 not found" is that correct?
If you are connected to the plug before the cable twist then the command would be 
"python gw.py read --drive=b --ecyl=39 mydisk1.scp"
Alternately keeping to a connection before a cable twist you can use a shugart command.
with DS0 set on drive command is "python gw.py read --drive=0 --ecyl=39 mydisk1.scp"
DS1 set on drive command is "python gw.py read --drive=1 --ecyl=39 mydisk1.scp"
DS3 set on drive command is "python gw.py read --drive=2 --ecyl=39 mydisk1.scp"

What are the results in each of these cases?

drdpj, 2020-01-27:
Drive select only works on the F7 variant with the custom PCB, so if you're
looking at a construction using the bluepill STM32F103 then the only way to
choose drive is either position it before/after the twist, or manually
configure the wiring. The question that needs to be answered first is what
sort of greaseweazle, if bluepill based is it using an additional PCB or is
it a direct wiring of a bluepill to the floppy drive?

If bluepill is wired as per the wiki, put the drive in front of any twist
on the cable, pin 12 and 16 are by default selecting the drive, and set the
drive ID to 1. A picture of your drive jumpers might help, and the actual
drive model number.

bradwh, 2020-01-27:
Thanks drdpj,
Sorry I forgot I am using a custom interface board that enables the selection of other drives.
So standard connection, then DS1 after the twist will be drive "a" default and before the twist should set to DS0 to be drive "a" default.
So Brainy78's Greaseweazle hardware should be good if the 1.2Mb drives work on HD disk ok.
Therefore pluging in a 360Kb drive into the same spot with the same DS selected should work OK unless the drives are faulty. However if I read correctly he has four different drives from two different manufactures all failing to read.
As you said a picture of the jumpers would be good and drive model numbers would also help.

brainy78, 2020-01-28:
@drdpj I use bluepill F1 which is wired as per the wiki.

I still cannot solve my problem.

HD FDD is ok but 2D FDD don't work. I just exchange each other with nothing changed.

[image]

My bluepill...
[image]

[image]

My HP FDDs and the Jumper spots...
[image]

[image]

[image]

[image]

And my Goldstar...
[image]

[image]

Is there some clues? T.T

drdpj, 2020-01-28:
If it is at the end of the cable (in other words, after the twist) the
drive must be set to the lowest value (1 on the HP and 0 on the goldstar)?
I will look at the other jumper settings a bit later!

andrewbird, 2020-01-28:
Do I see 150 & 330 Ohm resistor packs? I think @keirf mentioned above that it might be hard to drive a floppy without extra buffering if it has 150Ohm.

brainy78, 2020-01-28:
@andrewbird To be honest, I do don't know 150 & 330 Ohm resistor packs. Where could I find it? ㅠㅠ

andrewbird, 2020-01-28:
The resistor packs I'm talking about are on the floppy drives, they could be making the floppy impossible to drive from the BluePill directly. See the white 14 pin DIL on the Goldstar marked 899-3-R150.

brainy78, 2020-01-28:
@andrewbird Ah ha! I get it! Thank you.

Hp has a 330 ohm pascks and GoldStar has 150... Both of all could not drive from Blue. T.T

So I have no FDD to write right now...

keirf, 2020-01-28:
Try pulling the resistor pack.

bradwh, 2020-01-29:
What would be even better would be to change the termination resistor values with a replacement pack, but for a simple alternative we could use individual resistors.
Lets start with the GoldStar drive, pull the 14 Pin resistor pack (white top marked 899-3-R150) now fit in it's place 7x 1K ohm resistors R1 = Pin 1 to Pin 14, R2 = Pin 2 to pin 13, R3 Pin 3 to pin 12, R4 = Pin 4 to pin 11, R5 = pin 5 to pin 10, R6 = Pin 6 to pin 9, R7 pin 7 to pin 8.
This should bring the load within the capacity of the Bluepill.
You could also check the resistance of the termination on your working 1.2Mb drives between pin 10 and ground, you may find that these are even as high as 2K, in which case you could use 2k2 or 1k8 resistors instead of the 1K's but the lowest that will reliably work with the BluePill the better from line reflections point of view. 330 or 150 Ohm is just too much to ask of the BluePill.

brainy78, 2020-01-30:
@keirf Thank you for your kindness. I tried just removing resistor pack, but I saw same wrong result. I I think it need alternative termination resistor pack.

@bradwh Thank you your concern! ^^ I'll try as you suggest!! It looks difficult to me, but I'll do my best!

bradwh, 2020-02-07:
Hi brainy78,
Try this [https://kr.element14.com/bourns/4114r-1-102lf/resistor-network-isolated-7-resistor/dp/9355634](url)

closed 2020-03-11 by keirf