keirf/greaseweazle #17: Package missing in documentation: building from source

Acathla-fr, 2020-02-08:
Compilation fails with:
```
/usr/lib/gcc/arm-none-eabi/6.3.1/include/stdint.h:9:26: fatal error: stdint.h: No such file or directory
 # include_next <stdint.h>
                          ^
```
Which is a bit confusing.
Adding the package `libnewlib-arm-none-eabi` solves the problem.

keirf, 2020-02-08:
I've never had to refer to that package explicitly. Perhaps it's a missing dependency. What is the OS?

Acathla-fr, 2020-02-08:
Ubuntu 18.04
I just tried on another freshly installed laptop, same ubuntu, same "problem".

keirf, 2020-02-08:
The same distro I use, and I never saw or heard of this problem before. I will have a look what I have installed.

Acathla-fr, 2020-02-08:
I found the answer here looking for what `#include_new` was doing: 
`https://stackoverflow.com/questions/23973971/no-stdint-h-file-on-debian`

keirf, 2020-02-09:
On my box:
```
viper:~$ cat /etc/issue
Ubuntu 18.04 LTS \n \l

viper:~$ apt list --installed | grep arm-none-eabi

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

binutils-arm-none-eabi/bionic,now 2.27-9ubuntu1+9 amd64 [installed,automatic]
gcc-arm-none-eabi/bionic,now 15:6.3.1+svn253039-1build1 amd64 [installed]
libnewlib-arm-none-eabi/bionic,bionic,now 2.4.0.20160527-3 all [installed,upgradable to: 2.4.0.20160527-3ubuntu0.1]
libstdc++-arm-none-eabi-newlib/bionic,bionic,now 15:6.3.1+svn253039-1+10 all [installed,automatic]
viper:~$ apt-cache depends gcc-arm-none-eabi
gcc-arm-none-eabi
  Depends: libc6
  Depends: libgcc1
  Depends: libgmp10
  Depends: libisl15
  Depends: libmpc3
  Depends: libmpfr6
  Depends: libstdc++6
  Depends: zlib1g
  Depends: binutils-arm-none-eabi
  Recommends: libnewlib-arm-none-eabi
```

keirf, 2020-02-09:
So it is a Recommended dep of the gcc package. Those are supposed to be installed by default, which is why noone else has seen this issue on Ubuntu (nor for FlashFloppy, which uses same toolchain).

You might want to investigate why Recommended packages are not being installed. Looking for a mention of `Recommend` under /etc/apt/apt.conf.d/ would be a start.

Acathla-fr, 2020-02-09:
Okay. I found why: I install machines with PXE in automatic (preseed) mode and recommended packages where disabled, I don't remember why, probably because it was not activated a long time ago (before Lucid).

closed 2020-02-09 by Acathla-fr

keirf, 2020-02-09:
It's probably an Ubuntu packaging bug, to be fair: It's a full dependency if you can't even include stdint.h. But anyway, glad you got to the bottom of it!