keirf/greaseweazle #18: STM32 Bluepill Blinky test

bradwh, 2020-02-15:
I have sourced a couple of the STM32 Bluepill's from different sources and they have all failed the 2Hz Blinky test. All have communicated correctly at 115200 baud and seem to function in all other respects OK.

The problem I have is the blink rate itself. The documentation in STM32 fakes area states 
"Greaseweazle includes a Blinky test firmware in the alt/ subfolder. When written to a correct and working Blue Pill board, this firmware will flash the PC13 LED on and off at approximately 2Hz (ie. toggles every 500ms)."

Flashing on and off at 2Hz would give a toggle rate of 250mS not 500mS. At a toggle rate of 500mS would indicate that PC13 LED would be off for 500mS and then on for 500mS this would be a 1Hz flashing (10 flashes in ten seconds) which is precisely what I'm getting not the 2Hz (which would be 20 flashes in 10 seconds).
So what is correct? Should PC13 toggle it's state every 500mS or every 250mS?

Brad

keirf, 2020-02-15:
Changes state every 500ms.

keirf, 2020-02-15:
I will clarify docs to avoid hz reference and say instead changes state twice per second.

bradwh, 2020-02-15:
Thank you Keir for the clarification, I now know all my Bluepills are OK.

Regards Brad

closed 2020-02-15 by bradwh