keirf/greaseweazle #20: DENSel. Setting DD, HD density on 5.25" drives

nretro, 2020-03-03:
DENSel on Pin 2 of the floppy cable is an input for 5.25" drives and can be configured as input or output on some 3.5" drives. 
In the schematics for the F7_11 model DENSel is marked as an input (like INDEX). In the sources it is configured as an output, but I could not find a piece of code using that pin. Moreover it seems also that the protocol has no way of communicating DENSel.
My TEAC FD-55GFR refuses to read DD disks and does fine with HD ones. In theory I guess, one could just read a DD disk in HD mode and throw away every other track. But possibly the drive can do better than that if it knows that it is supposed to read a DD disk.
Is there any update in the pipeline or should I implement it myself?

keirf, 2020-03-03:
I will implement this as `gw.py pin 2 h` and `gw.py pin 2 l`. For HI=5v, LO=0v resp.

drdpj, 2020-03-03:
The density select pin won't make the drive double step. You'll still need
to erase every other track (HxC Floppy Software can delete "odd tracks" for
the entire side of the disk - this means you can do it very quickly). On
some drives pin 2 will make it spin at 300rpm instead of 360rpm. Most PCs
left the drive spinning at 360 and just dealt with the faster data rate in
the controller. It does potentially jiggle the coercivity setting for the
drive but that won't make much difference to reading, just to
writing.However, you shouldn't really be writing IBM 360k disks on a 1.2MB
drive as the head is too narrow, and they won't be reliable.

I use a TEAC 55 GFR with my greaseweazle, and it reads 360k disks fine
without needing to mess with the density setting.

nretro, 2020-03-03:
OK. Just checked the manual. @drdpj is right, the only effect of DENSel is the rotation speed and this could also be modified by setting the right jumpers on the 55 GFR. Interestingly, there is also a jumper to reverse the corresponding density of HI and LO on pin2.
So it would be a nice feature, since it's also in the schematics, but has no priority. Probably I will implement drive double step in the python code. Saves 50% of the processing time per disk. 
But that also means that my drive is out of alinement and I'll have to worry about that first. Or keep on using the 360k drive I've been borrowing from my sidecar.

commit 88b3af00 (2020-03-10): gw.py pin: Allow a pin to be manually set to specified level.
Fixes #20
closed 2020-03-10 by keirf