keirf/greaseweazle #23: Can't find serial

TzOk83, 2020-04-03:
Under Windows 7 x64 I can't access the Greaseweazle device. I get error:
```
File "D:\Program Files\Greaseweazle\scripts\greaseweazle\tools\util.py", line
71, in find_port
    raise serial.SerialException('Could not auto-detect Greaseweazle device')
serial.serialutil.SerialException: Could not auto-detect Greaseweazle device
```
Under Windows 10 x64 it works, but unless I try to use Zadig to install WinUSB (CDC) driver. It works only until is detected as USB Serial Device (COM9). When I use Zadig it is detected as Greaseweazle (COM9) but no longer works.

I'm using Python 3.7 (Anaconda).

keirf, 2020-04-04:
Are you using Greaseweazle v0.12?

TzOk83, 2020-04-04:
Ok, fale alarm, I didn't notice the v0.12 update, and I was running the v0.11. Now it works ok. Thank you!

P.S.
My Blue Pill has a CKS32F103C8T6 chip, and it seems to work ok.

keirf, 2020-04-04:
That fake chip does seem to work fine. No problem!

closed 2020-04-04 by keirf