keirf/greaseweazle #24: Same disk + different drives = different images

TzOk83, 2020-04-12:
I have tried to image same Atari ST disk with 3 different drives:
SONY MFP-F11W-5DU (SONY8903)
SONY MFP-F11W-U0 (SONY9002)
TEAC FD-235HF A110-U5 (TEAC 0302)

Two first drives are original DD drives from Atari ST computers, last one is a generic PC HD drive.
http://tzok.eu/ftp/mydisk.rar

bradwh, 2020-04-14:
And the question is?

keirf, 2020-04-14:
You won't ever get identical SCP dumps for successive dumps even using the *same* drive. Was that the question?

TzOk83, 2020-04-14:
Yes - tat was the question - is this expected/normal? How to assess image quality? Program itself didn't return any errors in any of these images, but one of them seems to be bad.

keirf, 2020-04-14:
You need to use analysis software to test/convert the SCP image. The SCP image dump just contains raw unchecked flux timings. So use HxC software (https://hxc2001.com/download/floppy_drive_emulator/HxCFloppyEmulator_soft_beta.zip) to visualise and convert.

Or on Linux use my Disk-Utilities project to convert (--format=ibm_mfm_dd): https://github.com/keirf/Disk-Utilities

Proper analysis in Greaseweazle itself is in development.

closed 2020-04-14 by keirf