keirf/greaseweazle #26: Writing problem of sector above INDEX

SuperTurboZ, 2020-04-20:
Greaseweazle is a good tool that works well. great.
The only problem is the writing end point.

Since the track write always ends on INDEX, the sectors above INDEX are broken.
So,the write end point should be the write noise that appears first from INDEX.
[image]

keirf, 2020-04-20:
This is true. Fixing it is a WIP (for example, I am currently introducing IPF write support and will include a fix for that image format). What is the source of the image you are writing here? Is it a raw dump that you are writing back? It looks like perhaps converting this to an IMG file and back to SCP may clean it up? (It would probably end up with tracks rotated to put the splice at the index though!). Something to be done in GW in future, but you could try doing these conversions via HxC software (load SCP, export as IMG, load raw, export as SCP). What do you think?

keirf, 2020-04-20:
(I ask about the ultimate source of this image or 'Master Data' as crossing the index like this, and no obvious larger track gap either, is pretty unusual. I guess this was constructed via some pretty special FORMAT program).

SuperTurboZ, 2020-04-21:
The image on the left is a game disc for SHAP X1 dumped with GW 0.12.
The image on the right is the SCP written in GW.

This is for sampling type copy protection.
Since the sector position is not checked, it can be handled by once converting it to D88 and adjusting it to the INDEX position.

However, there are other copy protections that cause  cossing INDEX, unstable bits and CRC errors combine.
This type of protection checks the contents of the sector data, so you need to avoid the splice points of INDEX.

If this problem can be cleared, it will be possible to bypass a large number of Japanese copy protections.
What still doesn't work is the non-magnetic protection and the multi-INDEX holes that physically punch holes.

SuperTurboZ, 2020-04-22:
In the meantime, would it be easy to manually offset the write end point(==INDEX position) in a parameter?

--index-offset={position/1 turn}

In this image, the write end point of cylinder 0 is not detected by HxC, so it will be difficult that the GW find the write end point automatically.
There are many such special formats in Japanese games.

[image]

keirf, 2020-04-28:
I shall look into it!

SuperTurboZ, 2020-04-30:
I wrote the first SCP image using gw0.14 and re-dumped it.
Some tracks have changed to readable, but the splice points still appear to be in INDEX.
Does "index misalignment" in the release notes mean something different from the crossing INDEX support?
[image]

keirf, 2020-04-30:
Index misalignment is supported only for writing of IPF image files. It is hard to find the write splice reliably in an SCP image file. Probably need to decode the SCP to sectors, remember all sector offsets, and then rebuild the track, or at least use the sector locations to intelligently pick a place to end the write. What do you think?

SuperTurboZ, 2020-04-30:
I didn't know IPF, so I was misunderstanding. sorry.

For data sector,

If there is no GAP at the INDEX position, I think it is better to shift the splice point to the GAP in the front or behind of the sector.

Alternatively, there is a way to detect where the bit rate is out of phase by +-5 to +-45%.

In the first step, it may be easier to detect the phase change.

For copy protect sector,

Since sectors that cross INDEX often overlap with other sectors in the middle of data, it is often better to shift the splice  point in front of the sector.

The other one checks the data in GAP. Placing a splice point on any sector boundary can break it.

It is not possible to support all tracks with one automatic detection, it is better to switch some search algorithms as an option.

SuperTurboZ, 2021-01-20:
I tried adding adittional 4ms data to Flux.splice in v0.24, but no additional data is written to media.
Does the current firmware of stm_f1 support parameter terminate_at_index=0 in usb.write_track?

closed 2021-12-31 by keirf

tdaede, 2022-11-16:
Do you know if, for these disks, the absolute sector ordering is important after the index, or just that the sector isn't corrupted by starting in the middle of it?

For example, is it okay to write the floppy with the index hole in between sectors instead, or do we have to make sure the index hole stays in the exact same spot, e.g. in the middle of the sector?

I'm hoping to do similar splice point detection (FM/MFM only) when I implement #229 .

keirf, 2022-11-16:
Probably most end up with sectors like this because the host doesn't care about INDEX for some reason. Or somehow the INDEX sensor was offset for some reason. I'm not sure if it's ever used for copy protection.