keirf/greaseweazle #27: Check for file existance when writing

dwillmore, 2020-04-27:
When writing a file, gw.py does not check to see if it exists before attempting to open it.  If it does not exist, this gives a long python error message which may not be very clear to non-programmers.

gw.py should check for the existance of the file to write and, if it is missing, display an error message to stderr.

andrewbird, 2020-04-27:
> gw.py does not check to see if it exists before attempting to open it.

No that's wrong, as it would introduce a race. What should happen is that the gw.py should attempt to open the file and handle any failure as necessary by displaying an error message.

dwillmore, 2020-04-27:
Ahh, yep, that's true.

keirf, 2020-04-28:
Fixed for next release (v0.14).

closed 2020-04-28 by keirf