keirf/greaseweazle #28: Erase doesn't let you specify the device/com port of the Greaseweazle to erase

M1kerochip, 2020-05-01:
Unlike the Read, Write, Reset and Update commands.

keirf, 2020-05-01:
Yes it does `gw.py erase <devicename>`

M1kerochip, 2020-05-01:
Ah, it doesn't like auto as the com port I think, then.
Works if I say COM9 but not auto.

keirf, 2020-05-01:
Are you sure? That code is identical to what is in the other commands, I just tried both `gw.py erase` and `gw.py erase auto` and both worked.

M1kerochip, 2020-05-01:
Hmm.
This is what I'm getting.
usage: gw.py erase [-h] [--drive DRIVE] [--scyl SCYL] [--ecyl ECYL]
                   [--single-sided]
                   [device]
gw.py erase: error: unrecognized arguments: auto

I'll download it again.

keirf, 2020-05-01:
Can you show me your exact command line that you are trying to run?

M1kerochip, 2020-05-01:
F:\gw\Greaseweazle-v0.14>python gw.py erase auto

F:\gw\Greaseweazle-v0.14\ is the path, and python has it's path statement set.

 I've been scratching my head on this all morning.

OK: Nevermind.
The new one I downloaded works perfectly.
BUT: only in a different directory. If I extract it into the same directory, it still gives an error.
A permissions problem?
I can't see anything special about this directory.

I don't see any problems with the physical disk in the event log.
Either way, working perfectly!

Actually, the old zip file works fine, too, in a new folder.
OK, marking as closed.

closed 2020-05-01 by M1kerochip

keirf, 2020-05-01:
That's weird XD By the way in any case you do not need to specify auto. You can just run `python gw.py erase` and sidestep the issue that way

M1kerochip, 2020-05-01:
It's a little concerning!!
I've dumped over 600 floppies using GW up to now, so, I'm reasonably used to it!! :D
It's really been driving me nuts all day.

I have a batch file set up and that small windows exe I wrote to dump too.
I just wanted to test the erase command before I add support in to the exe.
(exe, btw: https://github.com/M1kerochip/LaunchGreaseWeazle I'm sure I mentioned it before! ;)