keirf/greaseweazle #31: Epson SD-600 problem with STM32F103C8T6 Direct Connection

ILAHWWINC, 2020-05-10:
I want to use a "Blue Pill" with Dupont direct wiring on a Epson SD-600 5.25 Floppy drive.
I can read disks, but the disk analyze shows multiple errors (see dump1).

When i connect the same Epson drive with a STM32 "Blue Pill" adapter board and a floppy flat cable the result is fine (see dump2).

I can not find out what makes the difference.
[image]

[image]

keirf, 2020-05-10:
Does the adapter board have the 4x1k resistors attached?

ILAHWWINC, 2020-05-10:
Yes. It is one out of your production.

keirf, 2020-05-10:
Then either the dupont wires are not making good connection or, more likely, you need 1k pullups on inputs from the drive.

ILAHWWINC, 2020-05-10:
I tried with a 3.5 Floppy, Epson MPF-920-E. 

There is no problem. It seems to be a problem with the specific combination floppy-blue pill/dupont.

keirf, 2020-05-10:
Lack of 4x 1k pull ups on the inputs. Add them and it will work fine.

ILAHWWINC, 2020-05-11:
To be sure:
At „Blue Pill“ connect B6/INDEX, B7/TRK0, B8/WRPROT, B2/DKRD over resistor 1K to 5V. Right?

keirf, 2020-05-11:
Yes. Each gets a 1k resistor as you say. Four resistors total.

ILAHWWINC, 2020-05-12:
I can confirm that the problem is solved with the pull up resistors.
Maybe a note telling about the pull ups in the Hardware Assembly section would be helpful.

closed 2020-05-13 by ILAHWWINC

PicoElements, 2021-03-17:
In this special case with the SD600 , did you realized hot to slow down to 300 rpm ? Or you used  pin 2 L with success?