keirf/greaseweazle #32: Feature request: head stepper exercise mode

dwillmore, 2020-05-16:
I have a bunch of old drives (like everyone else) and they've been unused for decades.  I've cleaned them and greased the worm and sliders, but they are still stiff (probably junky grease hidden inside of the bearings).  If I move them manually--hard to do, messy, and possibly risky--I can get them to start seeking properly, but it's quite time consuming.

Would it be possible to add a head stepper exercise mode where it just steps the head to --ecyl and back to track 0?  Optionally a 'count' option for the # of times to do it (0 being forever).  Or I could just wrap it in a script if that doesn't fit into your vision of gw.py.

Thank you.

Obiwantje, 2020-05-16:
I second here - some tools for us refurbishing drives would be great indeed !

keirf, 2020-05-16:
Yes can do. I could also do a calibration mode? Read sectors off a known good Dos disk at certain tracks repeatedly and dynamically update #good sectors on screen in a loop?

dwillmore, 2020-05-16:
I wouldn't complain!  The more the better.  If you are going that far how
about seeking around and making sure the sectors are for the correct track?

keirf, 2020-05-16:
By the way if you do any scripting things like the head step is quite easy to knock up yourself. See how simple greaseweazle/tools/erase.py is for example, and a simple head step could be a few lines of code after importing greaseweazle.usb and maybe greaseweazle.tools.util

keirf, 2020-05-16:
Yes I would calibrate on the track you specify on the command line, reading both heads and reporting nr sectors are how many are track too high, low, or just right. In a loop til you ctrl-c. It's basically what I implemented in my amiga test kit.

closed 2021-12-31 by keirf