keirf/greaseweazle #37: Feature request: A2R support

d235j, 2020-05-28:
The A2R format is an unresolved format used by the Applesauce imaging hardware. Being able to read and write A2R would allow use of Greaseweazle with the Applesauce hardware or software.

The WOZ format is a resolved format also used by Applesauce. Supporting it would also be great, as many emulators support it.

The format specification for A2R is at https://applesaucefdc.com/a2r/.
The format specification for WOZ is at https://applesaucefdc.com/woz/.

An MIT-licensed Python package for handling A2R is at https://github.com/a2-4am/a2rchery.
An MIT-licensed Python package for handling WOZ is at https://github.com/a2-4am/wozardry.

d235j, 2020-05-29:
A2R may be a better unresolved flux format, because among other features it can store different numbers of reads per track. Perhaps it would be worth investigating A2R instead of SCP as the native flux image format.

d235j, 2020-05-29:
One limiting factor may be the 125ns resolution. Would be worth investigating if this is a problem and figuring out a proper solution for it.

abcbarryn, 2022-08-16:
Any updates on maybe having Woz format support?

d235j, 2023-02-23:
A2R v3 has made various improvements that allow it to be suitable for practically any unresolved or resolved flux image from any imaging device — the resolution is no longer fixed, and other changes have been made to facilitate this. The full specification is available at the same URL as above.

keirf, 2023-04-27:
Basic A2R decode is supported now by Greaseweazle. It doesn't support the multiple captures per track, just picks the one it likes best, for now. I will need to change the Image API to support advertising multiple captures per track.