keirf/greaseweazle #38: C64 Dump with GW 0.18 broken

ILAHWWINC, 2020-05-30:
I do a dump of a C64 disk with the default settings and GW 0.18. Then i try to convert the .scp to .flux with fluxengine. I get an „scp file corrupt“ error message.

Doing the same dump with GW  0.17 does not make any problems.

ourIThome, 2020-05-30:
I believe fluxengine software has issues with SCP files.
https://github.com/davidgiven/fluxengine/blob/master/doc/using.md

ILAHWWINC, 2020-05-30:
But when i use GW 0.17 firmware to dump, i can convert the scp with fluxengine. There must be a difference between GW0.18 and before.

ourIThome, 2020-05-30:
Oops sorry missed the last line about 0.17.

keirf, 2020-05-31:
Generation of single sided images has changed in v0.18 to match the supercard Pro author's intentions, and what he has explicitly stated in the latest spec. It also matches the behaviour of the latest beta hxc software (which has been similarly amended).

In short, fluxengine too probably needs to be fixed. Should be straightforward. I recommend to raise a ticket there. If you link it here I'll be happy to pipe up with the technical detail required.

keirf, 2020-05-31:
Another good thing especially if fluxengine author doesn't mind would be for me to add import and export of his flux files directly to Greaseweazle.

ILAHWWINC, 2020-05-31:
I opened a ticket at fluxengine github. I will report back.

ILAHWWINC, 2020-06-02:
I got no response from the fluxengine side so far.

When i try to use the gw.exe v0.17 together with the latest firmware 0.18 result is an error requesting firmware 0.17.  Will it be possible to „override“ / patch this?

keirf, 2020-06-02:
You can downgrade to v0.17 no problem. Just 'gw update' from the 0.17 tools.

keirf, 2020-06-08:
This is not yet fixed (v0.19). Should get it done for v0.20.

keirf, 2020-06-09:
Here is a version for you to try. On the command line you must explicitly ask for an "old-style" single-sided SCP image, for example like this:
```
gw read --single-sided --ecyl 40 a.scp::old_ss
```
(Note the `::_old_ss` tagged on the end of the image file name: That's the magic!)

[gw_38_1.zip](https://github.com/keirf/Greaseweazle/files/4754120/gw_38_1.zip)

ILAHWWINC, 2020-06-09:
Thanks alot. How should i use this, is this a different firmware, or v0.19. There is a file gw, no extension 3xx bytes.

keirf, 2020-06-09:
You can use the included firmware or v0.19 release. It's not a Windows exe so run it as "python gw..." on Windows.

ILAHWWINC, 2020-06-09:
Yupp, working as expected. Now i can do a one click dump to d64 with gw, fluxengine and a script. Thanks

ILAHWWINC, 2020-06-11:
rmueller83 I have problems with SCP software because of virus alert, cant try. But what happens when you patch to right id / commodore with hex editor? 

keirf I got response from hxc forum (https://torlus.com/floppy/forum/viewtopic.php?f=2&t=3865). May be a problem on gw side?

keirf, 2020-06-11:
Yeah I fixed it in the tools I attached here in an earlier post. The track signature error thing will be fixed in v0.20.

I will provide a cheesy way to set the platform type code in the scp header. Hexediting it will also work.

ILAHWWINC, 2020-06-18:
Here a discussion about the scp format and single sided disks. Maybe helpful for gw software too.

closed 2020-07-11 by ILAHWWINC