keirf/greaseweazle #40: Can't update v0.15 GW F1 board

dwillmore, 2020-06-03:
I have a (several actually) F1 BP Greaseweazle boards running v0.15.  I'm trying to get it to version 0.18, but I can't seem to get the update command to work.  If I use the gw.py from v0.15, it says:
`** Bootloader v0.15 [F1], Host Tools v0.15
../Greaseweazle-v0.16/Greaseweazle-F1-v0.16.hex: No match for hardware type 1`

The same goes for v0.17 and v0.18 .hex files.
If I try to use the v0.16 gw.py, I get:
`** Bootloader v0.15 [F1], Host Tools v0.16
** FATAL ERROR:
unpack requires a buffer of 2 bytes`

The same happens for the v0.17 and v0.18 gw commands with their respecitve .hex files.

Other than breaking out the st-link v2 device, what's the path forward here?  Am I doing something strange?  Machine here is Fedora F31 and GW otherwise works fine.

keirf, 2020-06-03:
I'll have a look. Possibly I've broken update files for F1.

keirf, 2020-06-04:
You have to use the host tools that match your update. For example use host tools v0.18 to update the firmware to v0.18.

dwillmore, 2020-06-05:
Using v0.18 to update to v0.18 fails the same way.

keirf, 2020-06-05:
Yes I can reproduce the problem. I will sort a fix.

keirf, 2020-06-05:
Oh no I can't. It works fine.

Your problem is that you should be specifying the .upd file, not the .hex file. In fact with latest tools, you shouldn't need to specify the .upd file at all -- it should be found auomatically.

dwillmore, 2020-06-05:
User too stupid to use tool.  Closing.

d235j, 2020-06-05:
Perhaps the error message should print out "Make sure you are using a valid .upd file" or something a bit less confusing?

keirf, 2020-06-05:
Yes that's a good idea. I was lazy and didn't put a proper header or checksum over the whole file.

commit 798a5db8 (2020-06-06): 'gw update': Add a file signature and overall CRC so we can validate
that we are passed a proper valid UPD file.
Fixes #40
closed 2020-06-06 by keirf