keirf/greaseweazle #41: SCP header issue with single-sided images

simonowen, 2020-06-07:
Single-sided .scp images created by the Greaseweazle tool seem to have half the expected values in the start_track and end_track header fields (offsets 6 and 7).

A 42-track image containing only the first side has 41 in the end_track field. Opening this image in v2.10 of the official SCP software (using Function->Analyzer/Editor, then Read button) shows only 21 tracks available when using the track slider in the bottom-right.

Hex-editing the image to change end_track to 81 (using a value in the full 0 to 165 range) gives the expected behaviour in the official SCP software. The specification is a bit unclear on how this value should be treated for single-sided images, but since the author's software seems to expect the full range value then I think it needs changing.

keirf, 2020-06-07:
Sigh. I did this with reference to Jim Drew's official proclamation in an HxC forum post here: 
https://torlus.com/floppy/forum/viewtopic.php?p=22514#p22514

Note the start/end values in Jim's stated "correct" output.

I will have a bit of an investigate...

simonowen, 2020-06-07:
Thanks Keir. It's strange the author's own tool doesn't seem to fit with his own example.

To me it would make more sense the values being in the full range, since that's how the index itself is stored. Though except for the potential to extend the size of the index they don't seem to really be of much use. I've just changed SAMdisk to mostly ignore them and scan the whole index of 166 entries, which is always present.

keirf, 2020-06-07:
Yes I quite agree. I was surprised by his concrete output example. I suspect it was contrived and buggy. I can confirm what you are seeing, and it's certainly safer anyway to have a value too big than too small. FWIW I'm ignoring the start/end fields on import anyway.

closed 2020-06-07 by simonowen