keirf/greaseweazle #44: Inverting output logic

staticmem, 2020-06-27:
I have a board that has output buffers that inverts the output states to the floppy. (the inputs are unchanged).

I've made this simple change in src/floppy.c:

```
/* staticmem: default GW code
#define write_pin(pin, level)
    gpio_write_pin(gpio_##pin, pin_##pin, level ? O_TRUE : O_FALSE)
*/

/* staticmem: inverted logic */
#define write_pin(pin, level) \
    gpio_write_pin(gpio_##pin, pin_##pin, level ? O_FALSE : O_TRUE)
```

I suspect though I also need some other changes?

keirf, 2020-06-27:
Change the definitions of O_TRUE and O_FALSE themselves. Should work.

staticmem, 2020-06-28:
Thanks,  I disabled the above change and edited the O_TRUE and O_FALSE as suggested.  Then I also had to change the pin configuration for the GPO_bus:

#define GPO_bus GPO_pushpull(IOSPD_LOW, HIGH)

keirf, 2020-06-28:
That's surprising. Are you patching against tip of master branch?

staticmem, 2020-06-29:
Not sure what you mean by "tip".

I used the Greaseweazle-0.19 source package and made changes to src/floppy.c and src/f1/floppy.c as needed then did a "make dist".

The changes involved the redefine of O_FALSE and O_TRUE,  changing some of the port lines to suit this boards requirements and also using #define GPO_bus GPO_pushpull(IOSPD_LOW, HIGH)
It was a while ago that I rearrange some of the lines and can't quite remember the full reason but making sure 5V tolerant was used for inputs and possibly a change or two to make routing the board out easier.

I also modified the PIN test code to test the new pin arrangement but I noticed in the background things like drives selection are disabled after a few seconds.

For output buffers I'm using 2N7002 NFETs.  These provide the required pull-down capability (I use a 150R res pack in last drive).  All the FETs will have have 10k gate pull down resistors fitted because on power-up of the BP it will be high impedance which can cause lines to be in any state until things settled down in the Micro.  My earlier board uses 10k pull downs on WG and WD only to prevent corruption of a disk if left in drive but my later board provides 10k for all the FET gates to avoid lines coming on at the start.

I also added support for 4 drives using this method.  I moved some of the floppy lines to suit the voltage requirements of my arrangement.  On an earlier design I also made changes for the extra drives and modified to the Python script to work like the F7 build which I will also do for this 2N7002 board.

I guess I could have set the direction to an input (with weak pull-downs) or high impedance when the output state needs to go to a high and set as output when a low is required.  Anyway I just went with the Push-Pull method as it will be quicker to respond given possible gate capacitance issues.

closed 2020-06-29 by keirf

staticmem, 2020-06-29:
Ok closed, that's fine as solution has been reached.  I sent you an email reply about my variation that I hope you will receive and consider.

keirf, 2020-06-29:
That's fine, though I have not received an email. If you have code to contribute then the best method is a Pull request. I can then give feedback. I don't recall if we discussed how to differentiate between different boards in a single build of Greaseweazle? We're using PC13-PC15 in the F7 build.

keirf, 2020-06-29:
Okay then email is best. But please tell me what you'd like done: Are you looking to have your board supported in the official releases? Or included as a build option in the source? Or what?

staticmem, 2020-06-29:
ok, sent you an email that works.