keirf/greaseweazle #45: Greaseweazle INI config

staticmem, 2020-07-06:
For HW variations and default settings it would be nice to have an optional INI style file.  If not present then use defaults as normal.  If present then apply the settings. i.e: default drive.  Interface HW variations.

keirf, 2020-07-06:
Yes this is a good idea.

bsdf, 2022-10-14:
+1 for this feature. one approach would be to replace `argparse` with [`configargparse`](https://github.com/bw2/ConfigArgParse), which claims to be a drop-in replacement. the library extends `argparse` to allow configuration settings to be read from a combination of config files, env vars, and commandline args

screwtop, 2024-11-01:
+1 from me, too - I had been thinking that env vars for drive and format at least would be very handy. If `configargparse` can do all of those easily, ideal?