keirf/greaseweazle #46: F1: Add ~2mm distance between power & data pins

DLehenbauer, 2020-07-19:
On the F1 board, the power and data pins are a little close for using a shrouded/keyed 34-pin data connector.

In my build, the key that protrudes inward on the FDD power cable catches the edge of the shroud.

With some bending you can force the power connection, but an additional 2mm of space between the power and data pins would give plenty of space to keep the two from interfering.

Here are the parts I used:
* [Shrouded 34-pin connector](https://www.digikey.com/product-detail/en/adam-tech/BHR-34-VUA/2057-BHR-34-VUA-ND/9832262)
* [Berg 4-pin connector](https://www.digikey.com/product-detail/en/te-connectivity-amp-connectors/171825-4/A106197-ND/2135511)

(Not a big deal, but thought I'd pass the info along in case you rev the F1.)

keirf, 2020-07-24:
Ah yes so I see... I could rev the design, but I don't know I'll personally be making any more.

I suppose obvious solutions are a right-angled Berg or use unshrouded 2-row pin headers.

keirf, 2020-07-24:
Are you thinking to get some more boards made yourself?

DLehenbauer, 2020-07-24:
Nope, no plans.  I'm just building one F1 for myself using a PCB I bought [from an eBay seller](https://www.ebay.com/itm/Greaseweazle-F1-PCB-for-PC/303619277862?hash=item46b11e8426:g:y~cAAOSwABBfB8Ap).

No worries.  Just an idea to keep in mind in case the opportunity arises.

DLehenbauer, 2020-07-24:
(Also, I didn't think of a right-angle Berg connector.  Thanks for mentioning it.  I'll pick one up next time I order parts and give it a try.)

closed 2020-08-06 by keirf