keirf/greaseweazle #47: GW 0.20, F7 hardware, "No Index"

drdpj, 2020-07-24:
Sony F920 HD 3.5" floppy drive, having to specify "drive 0" on a straight through cable otherwise I get "No Index"

keirf, 2020-07-24:
The drive-select jumper/bridge on the drive is presumably set for DS0 (pin 10)? Usually you can spot a jumper or solder bridge for this.

In which case `--drive 0` is correct. Now arguably most users would prefer a select-all kind of a default, similar to F1, as most users aren't hooking up multiple drives. Or perhaps allow a config file for default values like this.

drdpj, 2020-07-24:
Yes, it's set to D0 on that one - that's reassuring. Config file would be useful!

closed 2020-07-24 by keirf

vonj, 2023-03-10:
> The drive-select jumper/bridge on the drive is presumably set for DS0 (pin 10)? Usually you can spot a jumper or solder bridge for this.
>
> In which case `--drive 0` is correct. Now arguably most [...]

So on the F1 it doesn't matter if the cable is twisted or not? (For reading PC disks from PC drives.)

keirf, 2023-03-10:
That's right. The F1 adapter which the blue pill is soldered to has the wiring to support both ways.