keirf/greaseweazle #48: GW0.20 F7 Hardware, not reading properly

drdpj, 2020-07-24:
Sony F920, 3.5" HD drive (this does work fine with the F1 hardware), reads of high density and standard density disks are resulting in 8kb SCP files. See attached.
[test2.zip](https://github.com/keirf/Greaseweazle/files/4971149/test2.zip)

keirf, 2020-07-24:
```
$ python3 ~/Greaseweazle/scripts/misc/scp_info.py test2.scp | grep flux= | grep -v flux=0 | wc -l
0
```

keirf, 2020-07-24:
There are no flux values in the SCP file at all. Nothing was seen on RDATA.

This Greaseweazle hardware is working okay with other drive(s)?

keirf, 2020-07-24:
For a PC drive with a straight cable I would expect `--drive b` rather than `--drive 0`. Unless the drive-select configuration has been messed with on the drive PCB.

keirf, 2020-07-24:
Seems to me a few possibilities: Bad Greaseweazle, bad drive, bad cable, bad config.

How many can you discount?

drdpj, 2020-07-24:
It's bad GW I think, buzzed out OK but just looking through offspring's microscope I wasn't convinced by the soldering on the QFP.  Sorry - I was a bit quick off the mark there!

reinauer, 2020-08-07:
Is there a way to catch this in software? I ran into the same issue. If absolutely no data is coming, the error message should read "Go, resolder the STM32, you dummy" ;)

keirf, 2022-01-31:
Software these days displays the amount of flux read from each track. So `0 flux in 0.00ms` is probably clue enough.

closed 2022-01-31 by keirf