keirf/greaseweazle #49: Windows Python modules

staticmem, 2020-07-26:
I note since v0.16 there is a now a Greaseweazle WIN executable.  The Wiki Software-Installation page does not show the traditional method to install Python modules under Windows.  Before v0.16 I expect this was documented but the Wiki only reflects the latest release?

https://github.com/keirf/Greaseweazle/wiki/Software-Installation

Case in point is the installation of the "bitarray" Python module.  Can this be added to the Wiki?

staticmem, 2020-07-27:
Is it even possible to use the Python scripts under Windows or does it have to be a Windows executable?

I installed cx_Freeze and attempted a "make windist" but it failed.  I might open another issue for that but right now I just need to know what needs to be installed under Windows to run the Python scripts.  This used to work in earlier versions.

keirf, 2020-07-27:
Yes it can work in Windows from scripts.

keirf, 2020-07-27:
After all that is what cx_freeze freezes 😊

staticmem, 2020-07-27:
Going back to the original point what is needed to install the required Python modules under Windows please?

keirf, 2020-07-28:
If you have visual c compiler installed then all required modules can be installed via  python -m pip just as on Linux.

staticmem, 2020-07-29:
Unfortunately Keirf you now need propriety software to build the Windows bitarray unless you use one provided by a 3rd party.

python -m pip install crcmod pyserial bitarray
Fails to install "bitarray".

For others that may be interested here is a solution...
python --version

Make note of the version number.  My install is Python 3.8.5.
Go to this site where someone has built various bitarray 32 and 64 bit packages:
https://www.lfd.uci.edu/~gohlke/pythonlibs/#bitarray

You will see names that have "cp39", "cp28", etc.  These are the CPython version numbers.  You need to grab the one that matches your Python version and also 32 or 64 bit build.  For me I grabbed:
bitarray-1.4.2-cp38-cp38-win32.whl

Then run this using the pip command after setting the correct filename:
pip install bitarray-1.4.2-cp38-cp38-win32.whl

Now you should be able to run the install commands again:
python -m pip install crcmod pyserial bitarray

cd Greaseweazle-v0.20
python gw

You should see the GW options.

keirf, 2020-07-29:
What do you mean by proprietary software? The C compiler?

staticmem, 2020-07-29:
Yes, I see there is a "Community" version of it that features:
Powerful IDE, free for students, open-source contributors, and individuals.
Is it not possible to use a free open source solution such as mingw?

Not sure why there needs to be Python specific versions for the bitarray module.  It would be convenient if the WHL file could be distributed with GW.

keirf, 2020-07-30:
It doesn't matter, the binaries produced by the Microsoft compiler are not restricted license-wise. Bitarray and indeed python are already distributed for users in the windows prepackaged version of Greaseweazle. If you want to use the bare scripts for some reason, such as creating your own parallel releases, you have to deal with setting up the required environment yourself. It's not an installation that I support now I provide prepackaged binaries. This is to save me time.

staticmem, 2020-07-30:
Reason I use the Python scripts is because I made some changes to support by F1 compatible board interface which allows 4 drives and buffers and pull-up resistors that suits Shugart.  I needed to allow 4 drives but the GW F1 Python limits it to 3 drives.
I guess I could patch the EXE to allow 4 Shugart drives if it's just one byte.

ok, I understand you mainly targetting a Windows audience and it has to be kept simple for them.

So either I use Python scripts or set up a Windows build environment to use cx_Freeze or one of the others, or patch a byte or 2 in the EXE.  When I have more time I might look at the 2nd option or the user of my board understands only 3 Shugart drives with the current EXE unless you add something in as an INI config option I suggest elsewhere.

stuaxo, 2020-11-27:
@staticmem is it worth making a PR with these changes?

staticmem, 2020-11-28:
> @staticmem is it worth making a PR with these changes?

The Firmware changes I made are here, just follow the instructions to access the repository:
https://microbee-mspp.org.au/forum/viewtopic.php?p=16405#p16405

My Firmware changes are for v0.21 but I see Keirf has just released v0.22. So I will look at applying my patch for that release. Also keep in mind that I don't make an EXE release but as stated you can still use the Python scripts under Windows.

ps: I do make kits available as I'm so snowed under and only send kits to members in Australia anyway.  I probably should release the Gerber files for it.

closed 2020-12-26 by keirf