keirf/greaseweazle #50: Making an executable with windist

staticmem, 2020-07-27:
I've tried to create a Windows executable using the Makefile under Linux using:
make windist

First I installed "cx_Freeze":
python3 -m pip install cx_Freeze --upgrade

It reports:
cp -a Greaseweazle-v0.20/scripts/build/exe.win*/* Greaseweazle-v0.20/
cp: cannot stat 'Greaseweazle-v0.20/scripts/build/exe.win*/*': No such file or directory
Makefile:74: recipe for target 'windist' failed
make[1]: *** [windist] Error 1

Although it fails it appears to create a Linux gw binary file instead which appears to work (shows commands):
./Greaseweazle-v0.20/scripts/build/exe.linux-x86_64-3.6/gw

If I then edit the Makefile to read:
cp -a $(PROJ)-$(VER)/scripts/build/exe.linux-x86_64-3.6/* $(PROJ)-$(VER)/
(Removed the first * after 3.6) I get a bit further:

cp: cannot stat 'Greaseweazle-v0.20/lib/bitarray/VCRUNTIME140.DLL': No such file or directory

What else is required to build the Windows binary?  Having it build the Linux binary is nice though but need the EXE.

keirf, 2020-07-27:
You can't make the windows exe in Linux. It relies on running in a windows environment.

staticmem, 2020-07-27:
How about with Wine then or even a helper script to source the required DLLs?

keirf, 2020-07-28:
I don't know. I make windist on Windows so a cross environment is not something on my own to do list.

closed 2020-07-29 by staticmem