keirf/greaseweazle #52: MFM 40T DS DD SCP image to 80T 1.2Mb

staticmem, 2020-08-05:
As standard 40T DS DD disks use a data rate of 250 KHz can the SCP image of a SCP floppy disk created from a 40T drive be written back to an 80T 1.2Mb drive (with double step)?

As we know the standard HD 5.25" drives spin at 360 RPM where as the 40T at 300 RPM and that the floppy controller in PCs increase the clock rate when writing the data to compensate and creates the correct data rate.

Is there an option to write a 40T disk image read from a 80T 1.2Mb drive back to a standard 40T drive or could an option be added for it?

keirf, 2020-08-06:
It should just work, Greaseweazle adjusts bitcell timings to match rotation rate of the drive it is writing to.

Writing 40t with an 80t drive will be unreliable as the 80t doesn't write a wide enough track. I've only heard successful reports for reads, not writes.

staticmem, 2020-08-06:
ok, so reports I hear of it not writing is really an issue with narrow versus the wider track widths.  If find rare earth magnets are handy when moved over the disk jacket first helps to neutralise the existing data.

keirf, 2020-08-06:
That could help. Also perhaps for writes I should write the second track instead of skipping it?

staticmem, 2020-08-06:
It has helped me in the past.  I guess as an option writing the adjacent track would help but I think only if it receives an erase for that track otherwise the data can very because of minor rotation speeds even if side by side.

keirf, 2020-08-06:
Oh yes good point on the two tracks not being perfectly in sync. I think you're right it would probably hurt rather than help. Better to hard erase with a magnet and rely on one thin track.

staticmem, 2020-08-06:
Having an option to erase the track next to it might work though but should be done first.

keirf, 2020-08-06:
You can't beat erasing with a magnet in my experience

staticmem, 2020-08-06:
The reason I created this topic is that I have been told that my variation GW PCB will not write to a disk in a "true" 40T drive. (i.e: 300RPM and 250kHz data rate),  reads fine with 5.25" 40T and 3.5" 80T.  I have connected a cro to the WD and WG and see it changing but that was when writing to a 800 kB floppy in a 3.5" drive.  I will need to repeat that test again.

Do we know that true 40T drives can be written to?  I there anything the FW checks that could make this fail?  There are not that many 40T drives around these days.  I have purchased many 5.25" drives in recent years but only one of those is a true 40T drive with the others being 80T.  So I'm wondering how much testing was done on 40T drives.

keirf, 2020-08-06:
I see no reason it wouldn't work.

staticmem, 2020-08-06:
I agree and was always my thoughts,  but when I went to confirm this it does not overwrite the old data.

keirf, 2020-08-06:
No different than any other weazle. As you said, magnet erasure is your best bet.

staticmem, 2020-08-06:
If good erasure is required then I would have to suggest GW is sniffing the existing format before writing to it?  Would that be correct?

keirf, 2020-08-06:
Oh, sorry, I see you mean writing a 40T disk in 40T drive. That should of course work fine. It is not the Weazle's fault. Is the drive good, is the cable good, does the 40T drive need buffered outputs, etc etc

staticmem, 2020-08-06:
I will look into this further using my scope to see what is actually happening before I make any further comments but so far there are two users with the exact same problem.  We are both experienced in electronics and myself in software and we know the drives work as they have been tested on other hardware to be ok.

keirf, 2020-08-06:
If you're worried we can keep this open a while. But I can tell you Greaseweazle cares very little what drive it writes to. And as part of the write process it samples the drive rotation speed and adjusts the whole-track timing to 'fit'. See scripts/greaseweazle/tools/write.py, lines 33-37 and 56-58. Hence 300RPM vs 360RPM should really not be an issue.

keirf, 2020-08-06:
If you're writing to a true 40T drive at 300RPM then you should expect to see falling edges on write-data line at 4us and 8us intervals (FM), and also 6us if writing MFM (double density).

staticmem, 2020-08-06:
I will try to look at this again over the next few days.  It has me baffled at this stage and does not make any sense.

drdpj, 2020-08-06:
Just for the record I can successfully write disks on a true 40 track 5.25"
drive (300rmp), an 80 track double density 5.25" drive (300rpm) and an 80
track high density drive (360rpm) using both the F104 and F730
greaseweazles. It definitely works.

staticmem, 2020-08-07:
@drdpj:  Keirf has asked me to do some testing on the Index Pulse of the TEAC FD-55BR drive.  There is a spike of about 650ns on the IP line shortly after it goes low.  So it would seem the issue is related to some drives and yours probably does not have this issue.

staticmem, 2020-08-09:
Keirf sent me a patch that fixes the Index pulse spike issue from the TEAC FD-55BR drives and probably other models?
Hopefully it will become part of the main branch or a configurable option.

Volutar, 2020-08-15:
Does it relate to this issue? https://github.com/keirf/Greaseweazle/issues/7

staticmem, 2020-08-16:
@Volutar:  Possibly but Keirf is the best person answer.  Here are captures of what I found for my FIO2 GW board:
https://microbee-mspp.org.au/forum/viewtopic.php?f=47&t=2741

keirf, 2020-08-16:
Yes same issue as #7

closed 2021-01-16 by staticmem