keirf/greaseweazle #54: Feature request: Clean

ourIThome, 2020-08-12:
https://www.facebook.com/groups/greaseweazle/permalink/3198408880244711/

Seems some interest for a clean option.
Revolutions = 0
No output file

3.5" cleaning disks recommended for 30 second's.
Wonder if a prompt to re-apply isopropyl proceed / cancel? To try evenly wear on the disk?

keirf, 2020-08-12:
I'll come up with something plausible that takes a cleaning-time parameter. You can then run it more than once if you want to apply IPA to the disk more than once.

philpem, 2021-02-18:
I've implemented the Scrub method (from Dave Dunfield's ImageDisk) in DiscFerret: https://github.com/discferret/magpie/commit/071b74657977c6754dc9e6e64b2d10d8c24d6ad1

Please feel free to steal this code and adapt it for Greaseweazle, it works quite well!

keirf, 2021-02-19:
The scrub method is interesting. I'm not sure about the rapid back and forthing (cyls j+1, 0, 2j+1, j, 3j+1, ...). By the way magpie doesn't quite do this as it seeks to `cyl` rather than `a`.

I wonder if 0, j, 2j, 3j, ..., 3j, 2j, j, 0, j, ... back and forth would be suitable?

keirf, 2021-02-19:
Well, I'll stick with the zig-zag. See second commit above (bbee893). Looks okay?

philpem, 2021-02-19:
Looks pretty spot on to me. And I've fixed the bug in Magpie - whoops. Thanks for pointing that out.

philpem, 2021-02-19:
As for the algorithm - it looks like it aims to scrub the head across the disk to dislodge dirt. Changing directions might make that work better, and I think CopyQM uses a similar method.
But equally, I've seen tools which seek to (0, N, 0, N, ...) -- so who knows what "the best cleaning method" is?

closed 2021-02-21 by keirf