keirf/greaseweazle #55: Feature request: EDSK

SM7I, 2020-08-12:
Would live to see EDSK support to be able to write +3 / CPC disks.

staticmem, 2020-08-27:
I wrote a shell script to easily convert disks to different formats, visit this link:
https://www.microbee-mspp.org.au/public_repository/

Then navigate to "home/public/Projects/fio1_fio2_gw_scripts/fiox.sh" after completing the Google challenge.

It may also be run under Windows if "Gow" is installed.  Details are in the script.  Note "fio2" is my variation of Greaseweasel and "fio1" is an earlier Floppy design reader.  It needs some dependencies installed.

d235j, 2020-08-27:
@staticmem could you you put your software/scripts (not large downloads) somewhere where it's not behind a CAPTCHA?

staticmem, 2020-08-27:
> @staticmem could you you put your software/scripts (not large downloads) somewhere where it's not behind a CAPTCHA?

What's the problem with the CAPTCHA anyway?  Would you prefer it to go up on Facebook where you have to log in and have a Facebook account.

SM7I, 2020-09-03:
External script is a nogo. The support must be in firmware.

staticmem, 2020-09-04:
> External script is a nogo. The support must be in firmware.

It has nothing to do with Firmware, the firmware knows nothing about image formats except maybe SCP.  Once you have the SCP image it can be converted to whatever format you want using the appropriate software. SCP to EDSK conversion works and if you look at the script you can see how it works.

The link that was included before was broken but now working again.

keirf, 2021-02-26:
The initial write support can be tested using this automated build:
https://nightly.link/keirf/Greaseweazle/actions/artifacts/43511871.zip

staticmem, 2021-02-26:
Can I assume that the FW has not changed to make this happen and this EDSK support is handled in the Python scripts?

keirf, 2021-02-26:
Yep

staticmem, 2021-02-26:
Nice addition. :-)

merlinkv, 2021-05-10:
Hello Keirf.

Yesterday I successfully connected a 3 "drive to the Greaseweazle and can read / write a standard 3" disk with no problem. But I agree with @ SM7I and would greatly appreciate if EDSK support for reading / writing CPC / ZX discs were integrated into future firmware.

Thank you very much for your great work.

closed 2021-12-31 by keirf