keirf/greaseweazle #58: Imaging 40T Single Sided comes out as 80T

dandoore, 2020-10-14:
If I ask for 40T Double step and single sided I get an output file of 83 cylinders, this is reported by HxC tools and Samdisk (used here so I can paste it easier):

```
D:\Users\Dan\Documents\Greaseweazle>gw read --drive 0 --single-sided --double-step --ecyl 41 514-3-test.scp
** Greaseweazle v0.21 [F7], Host Tools v0.21
Reading Track 41.0...

D:\Users\Dan\Documents\Greaseweazle>samdisk info 514-3-test.scp
[514-3-test.scp]
 Type:   SCP
 Size:   83 Cyls, 1 Head

 application : SuperCard Pro software
       index : synchronised
        mode : read-only
     quality : preservation
         rpm : 300 rpm
         tpi : 96 tp
```

If I do the same without the single sided option it's fine in terms of cylinders (even though there is a junk second side)

```
D:\Users\Dan\Documents\Greaseweazle>gw read --drive 0 --double-step --ecyl 41 514-3-test.scp
** Greaseweazle v0.21 [F7], Host Tools v0.21
Reading Track 41.1...

D:\Users\Dan\Documents\Greaseweazle>samdisk info 514-3-test.scp
[514-3-test.scp]
 Type:   SCP
 Size:    42 Cyls, 2 Heads

 application : SuperCard Pro software
       index : synchronised
        mode : read-only
     quality : preservation
         rpm : 300 rpm
         tpi : 96 tpi

```

keirf, 2020-10-14:
This is confusion over how a single-sided SCP image should be laid out. Try with the latest HxC beta/snapshot software version. Or append ::legacy_ss to your scp filename, for example:
```
gw read --drive 0 --single-sided --double-step --ecyl 41 514-3-test.scp::legacy_ss
```

dandoore, 2020-10-15:
Yay!
```

D:\Users\Dan\Documents\Greaseweazle>gw read --drive 0 --single-sided --double-step --ecyl 41 514-4-test.scp::legacy_ss
** Greaseweazle v0.21 [F7], Host Tools v0.21
Reading Track 41.0...
SCP: Generated legacy single-sided image

D:\Users\Dan\Documents\Greaseweazle>samdisk info 514-4-test.scp
[514-4-test.scp]
 Type:   SCP
 Size:   42 Cyls, 1 Head

 application : SuperCard Pro software
       index : synchronised
        mode : read-only
     quality : preservation
         rpm : 300 rpm
         tpi : 96 tpi
```

Thanks Keir!

closed 2020-10-15 by dandoore