keirf/greaseweazle #59: Feature request: Support reading second side of single sided format

root42, 2020-10-21:
Right now gw supports the --single-sided option which will read from head 0. However lots of C64 floppies are flippy disks without a second index hole. So reading from an IBM double sided drive will result in "no index" errors when flipping the disk. 

Can we get an extended single-sided option to choose head 0 or 1?

root42, 2020-10-21:
...and already while reading about this I notice that the information would be read in in the wrong order. So that will be no good. And even if that was reversible in software, the tracks on the other side of the flippy disk will be offset. I guess this is not solvable in software?

root42, 2020-10-22:
I now learned that some other projects (I think Fluxengine is one) can ignore the index pulse for reading. Maybe this could be a solution. Not sure how this would be implemented in detail though.

d235j, 2020-10-22:
I would suggest ignoring the index and using a timer. However, an index is useful to have.

Applesauce adds a secondary index sensor that's wired to the board. This is probably the best way to go.

root42, 2020-10-22:
Yes, a hardware mod is possible. But having the option in software would be nice.

kgl2001, 2020-10-27:
It's still a valid request. I asked for the same on the Facebook page, and Keir has said he will implement:

Does anyone know if there is an option in GW to select either the upper or lower head on a double sided drive? I can see a --single-sided option, which forces read / write of side 0 only, but I can't see anything to force read / write of side 1 only. The BBC micro can access the sides individually. Confusingly, the BBC (DFS) convention is to refer to each side as a separate drive! So:
BBC drive 0 is GW drive 0, side 0 and BBC drive 2 is GW drive 0, side 1 
BBC drive 1 is GW drive 1, side 0 and BBC drive 3 is GW drive 1, side 1

keirf, 2020-12-20:
This should be sorted as of next release. The new --tracks= syntax is documented in the wiki and available in automated builds (look under the Actions tab at the top of the Greaseweazle Github page).

closed 2020-12-20 by keirf