keirf/greaseweazle #62: Wiki update: Firmware install from Mac

dougep, 2020-12-02:
Hi, it would be useful if you could add the following to your wiki [Firmware Programming](https://github.com/keirf/Greaseweazle/wiki/Firmware-Programming) page alongside the Windows instructions:

**Mac:**
- Download the firmware zip file from [here](Download).
- Install [stlink](https://github.com/stlink-org/stlink): `brew install stlink`
- Connect your ST-Link programmer (**not** Greaseweazle!) to your Mac via USB.
- Identify the correct HEX file for your board (**F1** HEX file for Blue Pill, **F7** HEX file for F7 boards).
- Write the file: `st-flash --reset --format ihex filename`
- Ensure no errors were reported

dougep, 2020-12-02:
Should actually be identical on Linux as well, but I haven't tested

keirf, 2020-12-20:
Done, thanks!

closed 2020-12-20 by keirf