keirf/greaseweazle #68: Amstrad 3" Drive Requires GW Reset After Each Disk

trevcjohnson, 2020-12-29:
I'm finding I have to run a GW reset after every disk to get the drive to seek back the track 0 after transferring a disk.  If I don't it starts from where it last finished and then hits the stops.  I sold find that after running GW reset it seems to over seek when returning to track 0 like it keep trying to move but can't for a split second.  I have wired my drive for DS1 (Drive B)

keirf, 2020-12-29:
Are you using a tested cable/adapter to connect to the 26-pin Amstrad drive? It sounds like there is some kind of problem with the /TRK0 signal. When seeking to track 0 the Greaseweazle will seek outward until that signal is asserted. If the signal looks already asserted, then no seeking will occur and your next read will continue further inwards! Quite why it would then decide to seek properly on/after reset I'm not sure. But if it's then over-seeking... I would guess that the TRK0 signal is not well connected. Perhaps it floats?

trevcjohnson, 2020-12-29:
I did a quick continuity check and it appears to have continuity on the cable but you are probably right as it doesn't seem that happy .  I have another drive in a PCW I can pull later and try it.

keirf, 2020-12-29:
That would be good. No 3.5" drives?

trevcjohnson, 2020-12-29:
I just pulled the A drive out my PCW.  This one behaves differently, it fails with "Command Failed: Seek: Track 0 not found".  The drive definately seems to try to move back to track 0.  I am using the same cable and it has continuity between the pin 18 on the greaseweazle F7 v2 and the drive pin.  I have swapped back to the other drive and this is working albeit with the original issue.

keirf, 2020-12-29:
Maybe try increasing the seek time? Like `gw delays --step=10000`

keirf, 2020-12-29:
Also, trk0 signal is pin 26 at greaseweazle end. And dome three inch drives may need 12v power and/or more current. Perhaps you could externally power the drive?

trevcjohnson, 2020-12-29:
Sorry yes TRK0 is pin18 on the Amstrad drive.  I think the power supply is the issue, I've got the drive powered up inside the PCW and the data cable connected to the GW and this is now reading discs but difficult to tell if the trk seeking issue has gone as the drive is back in the metal shroud but seems promising.  Possibly the voltage is too low on the 5v.

keirf, 2020-12-29:
Not surprising really. Only 3.5 drives generally seem reliable on usb power and even that's not universal

keirf, 2020-12-29:
You accounted for 5v and 12v pins being swapped on amstrad power connector?

trevcjohnson, 2020-12-29:
It was definately the power supply on the 5v.  I made a little adapter to power it from a single 12v PSU (1.5a) with a step down module to 5v, reversed as you say, I reflowed the connections on this little board and tweaked the voltage up a touch, it now seems to work flawlessly.  Thanks for the advice.

keirf, 2020-12-30:
I should mention power requirements in the wiki. I'll do that then close the ticket.

keirf, 2021-01-15:
Done.

closed 2021-01-15 by keirf