keirf/greaseweazle #69: GW 0.23 doesn't seem to be working on an F1 bluepill.

M1kerochip, 2021-01-13:
I'm using a Windows 10 machine, just fyi!

I tried updating as normal, but the upd file wouldn't flash. Just sat there forever.
When I end tasked the gw.exe it said:

** FATAL ERROR:
unpack requires a buffer of 2 bytes

So, I manually flashed the .hex files. (To two different F1 bluepills, the F1 'Blackpill' from RobotDYN and a generic F1 Bluepill)

Both bluepills look similar. (I'm using an F1 PCB board, too, with resistors soldered on, which worked fine up to yesterday, so I assume it's not that)

Todays disk dump:
[image]

Previous disk dump, same drive, same disk:
[image]

I tried dumping the same disk with all 3 drives I use, using 2 different bluepills.
Same results, more or less.

If I dump a Kryoflux image, (rather than a SPC) trying to open it in HxC tools crashes the program.
The images are .scp files, generated from Aufit.

https://mega.nz/file/xDADXIzR#H7W3lEER-TjPUMbk1o2peaxI63JwrWUa1hZRAbpp2pY

Disk images:

Previous disk dump: CaptainDynamo_Codemasters_Disk1of1_AtariST_Dump1.scp

Then, drive 1(worst drive), 2, 3 and drive 3 with bluepill 2.
Drive 1 and 2 are PC drives, drive 3 is a Teac FD-235F Amiga 500 drive.

Dumps created using:

gw read CaptainDynamo_DefaultsDrive3_Bluepill2.scp

I initially tried: 
 read --tracks="c=0-83:h=0-1:step=1" --revs=5 CaptainDynamo_Codemasters_Disk1Of1_AtariST_Dump1.scp

But that gave really weird looking results.

etc (ie defaults)

Also, --revs=3 produces images with 4 revolutions, in previous versions it produced images with 3.

keirf, 2021-01-14:
I can't reproduce the error on update. Maybe it's just a rare blip. It should not be necessary to reprogram from HEX as the bootloader should not get overwritten. You could have tried reconnecting the Blue Pill and retry the command.

The problem with SCP/KF files and HxC seems to be a bug in HxC. It doesn't like stream files that don't start at the index mark. If I force index-cueing in the generated files then they load into HxC just fine. If I don't, my SCP output looks rather like yours. And my KF files fail to load (rather than crashing) but it's surely the same issue you've seen.

M1kerochip, 2021-01-14:
Yeah, not too worried about the update.

It happened once before on one of the devices, but I couldn't reproduce it since then.
Just said I'd mention it.

If it's a bug in HxC that doesn't like starting at the index mark, then most of the other programs that use .scp images are the same way.
The .scp files from 0.23 won't load in an emulator, either.
Not the Amiga ones, or Atari ST ones.
Neither Aufit nor HxC can produce working .adf / .st / .stx files from the images too.
So, they're not much use as they are presently.
(I can't test in a real machine at the moment)

Maybe index-cuing should be the new default?

keirf, 2021-01-14:
Let me do a quick test build for you to try.

keirf, 2021-01-14:
Please try the following build. No need to update the firmware, just run the new host tools. KF and SCP images are index-cued:
https://nightly.link/keirf/Greaseweazle/actions/artifacts/35243779.zip

M1kerochip, 2021-01-14:
Yeah.
Seem to be fine.

Dumped with: 
 read --tracks="c=0-83:h=0-1:step=1" --revs=5 "CaptainDynamo_Codemasters_Disk1Of1_AtariST_Dump1.scp" 

[image]
[image]

keirf, 2021-01-14:
Great thanks. Can you try KF too. And also SCP in your emulator and aufit when you have a chance?

M1kerochip, 2021-01-14:
Both the Atari and Amiga images (.scp and .stx/.adf etc) work in emulators, too.

(SteemSSE and WinUAE, both latest stable, and latest betas)

HxC seems to be able to load and process them, too.

And, the KryoFlux dump:
[image]

Also looks perfect in HxC and can export fine. So.

All looks good.

keirf, 2021-01-14:
Excellent thanks

commit 48f165a8 (2021-01-14): gw: Index-cue generated SCP and Kryoflux files.
Almost all external tools and emulators seem to need this.
Fixes #69
closed 2021-01-14 by keirf