keirf/greaseweazle #70: My custom patch fails on 0.23 and 0.24

staticmem, 2021-01-15:
The patch file I created for the F1 version 0.19 release for my custom board is no longer able to patch against the 0.23 and 0.24 releases.  It works against 0.19, 0.20, 0.21 and 0.22.

Below is the output when applying my patch.  It's having issues with the F1 floppy.c patch "src/f1/floppy.c", "scripts/greaseweazle/tools/read.py" and "patching file scripts/greaseweazle"/tools/write.py

```
../gw-patch.sh ~/Downloads/Greaseweazle/Greaseweazle-0.23.tar.gz
patching file src/floppy.c
Hunk #1 succeeded at 34 with fuzz 2 (offset 2 lines).
Hunk #2 succeeded at 267 (offset 43 lines).
patching file src/f1/floppy.c
Hunk #2 FAILED at 66.
Hunk #5 succeeded at 307 with fuzz 1.
1 out of 5 hunks FAILED -- saving rejects to file src/f1/floppy.c.rej
patching file scripts/greaseweazle/tools/util.py
Hunk #1 succeeded at 40 (offset -2 lines).
patching file scripts/greaseweazle/tools/read.py
Hunk #1 FAILED at 74.
1 out of 1 hunk FAILED -- saving rejects to file scripts/greaseweazle/tools/read.py.rej
patching file scripts/greaseweazle/tools/write.py
Hunk #1 FAILED at 67.
1 out of 1 hunk FAILED -- saving rejects to file scripts/greaseweazle/tools/write.py.rej
patching file Rules.mk
```

The end result is a failure to compile my F1 Firmware changes, below is an example of some of the compile errors.  All errors are to do with #define pin_* and #define gpio_*.

```
In file included from floppy.c:53:0:
f1/floppy.c:70:0: error: "pin_wrprot" redefined [-Werror]
 #define pin_wrprot  8 /* PB8 */
 
f1/floppy.c:19:0: note: this is the location of the previous definition
 #define pin_wrprot  15 /* PB15 */
 
f1/floppy.c:73:0: error: "gpio_densel" redefined [-Werror]
 #define gpio_densel gpiob
 
f1/floppy.c:33:0: note: this is the location of the previous definition
 #define gpio_densel gpio_sel3

```
I'm not sure how to proceed from here.  Maybe I need to pass the patch command some options to get around this?  I'm trying to avoid having to create a new patch from scratch.

keirf, 2021-01-15:
Point me at the patch and I'll redo it.

staticmem, 2021-01-15:
https://www.microbee-mspp.org.au/repository/
- Satisfy the rePATCHA check.
- Login to the repository as user "pub" and use password "pub"
- Browse to "Projects/fio2/patches"
- Download "gw-patch.sh" (don't use the  "gw-patch-2.sh") file.
  
Run the script as is and it will give usage information.  The actual patch file is part of the script that is redirected to a patch file then applied from the main script.

keirf, 2021-01-15:
Attached new patch.

[gw-fio2-revised.zip](https://github.com/keirf/Greaseweazle/files/5820086/gw-fio2-revised.zip)

staticmem, 2021-01-15:
Thanks for that I will check it out.

staticmem, 2021-01-16:
The modified patch seems to work fine and I can build 0.23 and 0.24 now.  I've not updated the FW at this stage
Testing the 0.24 Python script I run:

```
$ ./Greaseweazle-v0.24/gw --help
Usage: ./Greaseweazle-v0.24/gw [action] [-h] ...
  -h, --help  Show help message for specified action
Actions:
*** WARNING: Optimised data routines not found: Run scripts/setup.sh
  info        Display information about the Greaseweazle setup.
  read        Read a disk to the specified image file.
  write       Write a disk from the specified image file.
  erase       Erase a disk.
  seek        Seek to the specified cylinder.
  delays      Display (and optionally modify) Greaseweazle drive-delay parameters.
  update      Update the Greaseweazle device firmware to current version.
  pin         Change the setting of a user-modifiable interface pin.
  reset       Reset the Greaseweazle device to power-on default state.
  bandwidth   Report the available USB bandwidth for the Greaseweazle device.
```
And see the WARNING message so I run scripts/setup.sh:

```
$ ./Greaseweazle-v0.24/scripts/setup.sh
Collecting bitarray
  Using cached bitarray-1.6.1.tar.gz (55 kB)
Processing /home/staticmem/.cache/pip/wheels/50/24/4d/4580ca4a299f1ad6fd63443e6e584cb21e9a07988e4aa8daac/crcmod-1.7-cp36-cp36m-linux_x86_64.whl
Collecting pyserial
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Using legacy 'setup.py install' for bitarray, since package 'wheel' is not installed.
Installing collected packages: bitarray, crcmod, pyserial
    Running setup.py install for bitarray ... done
  WARNING: The scripts pyserial-miniterm and pyserial-ports are installed in '/home/staticmem/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed bitarray-1.6.1 crcmod-1.7 pyserial-3.5
WARNING: You are using pip version 20.2.3; however, version 20.3.3 is available.
You should consider upgrading via the '/usr/bin/python3 -m pip install --upgrade pip' command.
running install
running build
running build_ext
running install_lib
running install_egg_info
Removing ../greaseweazle/optimised/optimised-0.0.0.egg-info
Writing ../greaseweazle/optimised/optimised-0.0.0.egg-info
```

Running  ./Greaseweazle-v0.24/gw --help again still reports *** WARNING: Optimised data routines not found: Run scripts/setup.sh

I assume the setup script is something new since 0.22?

keirf, 2021-01-16:
Yes it is new in v0.23 and later. It provides C-optimised versions of hot conversion routines.

What system are you running on (eg. Ubuntu Linux, macOS, ...)? What if you run the script within the Greaseweazle-v0.24 folder?
```
$ ./scripts/setup.sh
```

staticmem, 2021-01-16:
I'm running Linuxmint.

ok, now I CD to this directory...

patched_0.24/Greaseweazle-0.24-fio2/Greaseweazle-v0.24

And try again, now it seems to work as expected...

```
$ ./scripts/setup.sh 
Requirement already satisfied: bitarray in /home/stewart/.local/lib/python3.6/site-packages (1.6.1)
Requirement already satisfied: crcmod in /home/stewart/.local/lib/python3.6/site-packages (1.7)
Requirement already satisfied: pyserial in /home/stewart/.local/lib/python3.6/site-packages (3.5)
WARNING: You are using pip version 20.2.3; however, version 20.3.3 is available.
You should consider upgrading via the '/usr/bin/python3 -m pip install --upgrade pip' command.
running install
running build
running build_ext
building 'optimised' extension
creating build
creating build/temp.linux-x86_64-3.6
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.6m -c optimised.c -o build/temp.linux-x86_64-3.6/optimised.o
creating build/lib.linux-x86_64-3.6
x86_64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-x86_64-3.6/optimised.o -o build/lib.linux-x86_64-3.6/optimised.cpython-36m-x86_64-linux-gnu.so
running install_lib
copying build/lib.linux-x86_64-3.6/optimised.cpython-36m-x86_64-linux-gnu.so -> ../greaseweazle/optimised
running install_egg_info
Writing ../greaseweazle/optimised/optimised-0.0.0.egg-info

```
And the GW command now appears to work too.

```
$ ./gw --help
Usage: ./gw [action] [-h] ...
  -h, --help  Show help message for specified action
Actions:
  info        Display information about the Greaseweazle setup.
  read        Read a disk to the specified image file.
  write       Write a disk from the specified image file.
  erase       Erase a disk.
  seek        Seek to the specified cylinder.
  delays      Display (and optionally modify) Greaseweazle drive-delay parameters.
  update      Update the Greaseweazle device firmware to current version.
  pin         Change the setting of a user-modifiable interface pin.
  reset       Reset the Greaseweazle device to power-on default state.
  bandwidth   Report the available USB bandwidth for the Greaseweazle device.
```
So now I need to update to the new customised FW and test that it works.

staticmem, 2021-01-20:
I updated the FW but it's not working.  It would seem to be using the original code without the patch having done anything except to change the versions reported?

I see you do this type of thing in the revised patched but I think I must be applying in incorrectly so it does nothing.  Also the "diff" command under my Linuxmint system does not recognise the "--git" option:

diff --git a/Rules.mk b/Rules.mk

keirf, 2021-01-20:
If you apply the patch within a git repository, do the following to see if it is applied: `git diff`

The line I use to apply the patch is: `patch -p1 <my.patch` where my.patch is the patch filename

keirf, 2021-01-20:
I have fixed the setup.sh script to check that it is run from the correct folder, and fail otherwise.

staticmem, 2021-01-20:
ok, how do I patch within a git repository?  I'm not used to using Git,  I've just been patching up against the extracted source TAR of the version to be patched but may be I should do it against the Git?

keirf, 2021-01-20:
No, you can patch within the untarred ball instead. Same `patch` command line should apply. You will have to manually check if the patch really applied though as `git diff` of course won't work.

The comment on setup.sh was relating to your earlier failure running that script. It is just to track that I have patched up that issue now. It is committed to the Git repository. But it doesn't affect what you're trying to do right now at all.

staticmem, 2021-01-20:
ok, I must be calling it up incorrectly then as nothing seems to get patched.  I'll check it out later.

closed 2021-01-27 by keirf