keirf/greaseweazle #72: F1 "bluepill": 0.24 (and 0.23) fail to write disk. (0.22 has problems, 0.21 and below work perfectly)

M1kerochip, 2021-01-22:
Hi,

I've done some testing, and it looks like the last version of GW that writes a whole disk for me, is 0.21.
Tried multiple drives, multiple STM32F1 "bluepill" devices.
(Generic, STM32F103 from Robotdyn, APM32F103 from Robotdyn)

At random stages, but on earlier tracks usually, 0.22 fails to complete writing a disk image.

I've tried writing the same disk image, from 0.17-0.21 and it wrote 5 times in a row.
(and multiple times, just to be sure, on 0.21)

With 0.22, I randomly get:

First time:
F:\gw\Greaseweazle-v0.22>gw write D:\Programming\Projects\sdbox-master\IDE-Setup_adf.scp
Writing Track 11.1...Command Failed: WriteFlux: Disk is Write Protected

the second time:
F:\gw\Greaseweazle-v0.22>gw write D:\Programming\Projects\sdbox-master\IDE-Setup_adf.scp
Writing Track 12.0...Command Failed: WriteFlux: Disk is Write Protected

Third time:
F:\gw\Greaseweazle-v0.22>gw write D:\Programming\Projects\sdbox-master\IDE-Setup_adf.scp
Writing Track 22.1...Command Failed: WriteFlux: Disk is Write Protected

So, trying to write this disk, you can see, first it failed on track 11, then 12, then 22.
On the fourth time, it was an earlier track. The farthest I got writing was the third attempt.

It makes no difference what physical disk/disk image combination I try, results are similar each time.

For 0.24, before it tries to write track 0.0, I get the write protect message.

This is the disk image I tried, 
https://modelrail.otenko.com/assets/amiga500/IDE-Setup.adf
It makes no difference if it's an .adf, .hfe or .scp, results are the same.

keirf, 2021-01-23:
Are you using dupont wires or a greaseweazle adapter pcb for connecting the blue pill boards?

M1kerochip, 2021-01-23:
Sorry, I should have said: It's a PCB with resistors soldered on.

And I've reflashed each version from 0.16-0.24 to test.

keirf, 2021-01-23:
I would still suspect bad or loose cable despite it seeming to be version related. But I will test 0.24 on my bluepill later.

M1kerochip, 2021-01-23:
I've tried different PCBs, different cables, different STM32F1 boards.

With each combination, each time, the 0.22-0.24 versions fail to write.

Re-flash 0.20 and the same setup writes perfectly.

I've tried making a new cable, only 6cm, and it too works perfectly on all versions up to 0.21

I've tried writing a disk image more than 30+ times in the last day or so, the only consistent fails are the 0.22+ versions.

Atm, I have 2x PCBs, and 2x F1 boards. If I swap them, the 0.24 board fails, no matter which cable/drive it's connected to.

It's unlikely to be a cable issue, I think.

keirf, 2021-01-23:
Dodgy write-protect switch in the drive perhaps? That is directly tested at the start of each track write, and that is the only place which returns ACK_WRPROT (which gets the write-protected error string in the gw application).

[image]

keirf, 2021-01-23:
Also maybe ask more widely on the FB group. There are a lot of F1 users, and some proportion will certainly have updated to v0.22+ and attempted to write a disk.

Maybe also check pull-up resistance on the adapter board for floppy pin 26.

M1kerochip, 2021-01-23:
I'm not a FB user unfortunately.

I don't think it's a drive issue, all 8 of my drives (3x PC, one new old stock, 2x Amiga, 3x Atari ST) behave exactly the same.

All of my F1 boards are either from RobotDYN, (ATM32F1 or APM32F1) or one other Chinese manufacturer, so, I definitely don't have a lot of variety there though.

I'm writing on Windows, too, if that makes a difference.

I'll try a dupont connection just for the hell of it.

keirf, 2021-01-23:
Just tested v0.24 on Windows 10 and it works okay for me. I have posted on the FB group about this, linking to this issue ticket.

keirf, 2021-01-23:
Which country are you in?

M1kerochip, 2021-01-23:
Ireland!

keirf, 2021-01-23:
Well I can only suggest send to me for test. I'm UK. Others on the FB group are going to test also.

M1kerochip, 2021-01-23:
Good Stuff.

I'm out for a few hours, but later I'll try the duponts, and a different USB cable, too, just in case, and I'll dig out my laptop, and try on that, as well.

ourIThome, 2021-01-23:
With the randomness in progress of failing ... Can I ask what spec your computer is?  How busy is your computer.?

M1kerochip, 2021-01-23:
Sure: Win10 Pro, 16GB, i7-4790k, Samsung EVO 860 Pro.

Not a new machine, but, it isn't even vaguely stressed by this.

Most of the time, the CPU usage is at a few percent.
The machine isn't doing anything else.
And, like I said, a quick flash back to 0.20, and it writes fine.

It's not exactly random, either, 90% of the time it fails on track 10.0-12.1.
100% of the time, on 0.24, it fails on track 0.0

And, it's never failed to read a disk. (1,000+ disk dumps so far)

I've tried 4-5 more random disks, all failed to write on 0.22.
Flash back to 0.20, all wrote fine.
(I've tried writing the .upd file, and flashing the .hex with an ST-Link, neither of which seemed to make any difference)

gw bandwidth says it needs ~ 8.2mbps.

                   Min.   /   Mean   /   Max.   
Write Bandwidth:    9.136 /    9.206 /    9.230 Mbps
Read Bandwidth:     9.038 /    9.553 /    9.734 Mbps

Estimated Consistent Min. Bandwidth: 8.134 Mbps
 -> Max. Flux Rate: 0.915 Msamples/sec
 -> Min. Ave. Flux: 1.093 us

ourIThome, 2021-01-23:
I know with my slow laptop, if I try to capture to scp ... Then open outlook, chrome it stops the capture.  So think that rules that out.

Someone asked about your python version ... Can I assume your using the following file in its own clean folder?

https://github.com/keirf/Greaseweazle/releases/download/v0.24/Greaseweazle-v0.24-win.zip ... And that your running gw.exe?

I'll see if I can find a blank disk to try and write to tomorrow.

ourIThome, 2021-01-23:
I had another thought ...

I believe GW has an erase option ... Are you able to complete this fine?

And ... Is (which doesn't make sense to me) the floppy disk a high density disk (two square )... Where the one without the tab to write enable is missing from the opposite side?  If so ... Can you cover the hole up?

Have you tried writing a normal PC image to disk?

M1kerochip, 2021-01-23:
Yeah, each version I've tried is in it's own folder.

The 0.24 blackpill: (obviously, the version depends on what's flashed)

Host Tools: v0.24
Greaseweazle:
  Device:   COM9
  Model:    F1
  Firmware: v0.24
  Serial:   GW0106617A53471310004E4C38
  USB Rate: Full Speed (12 Mbit/s)

STM32 Blackpill : https://robotdyn.com/stm32-arm-arduino-mini-system-dev-board-blue-pill-with-arduino-bootloader.html

APM32 Blackpill: https://robotdyn.com/black-pill-apm32f103cb-128kb-flash-20kb-sram-stm32-compatible-arm-cortexr-m3-mcu-mini-board.html

Datasheets and pinouts are on respective pages.

I have a generic bluepill too.

All three devices work fine on 0.20, any combination of cables, pcbs, drives, and disks, and all three fail on any combination of the 8 floppy drives, and physical disks. on 0.24.

I have two PCBs, 10-20 floppy cables (Atari/Amiga striaght through, and PC twisted etc) and about 10 different F1 boards.

All disks are DD. All disk images are standard amiga DD disk images I'm trying to write.
I mostly use the TEAC Amiga 500 DD drive, since I mostly dump DD disks.

ERASE finishes out fine, on each version. 0.20 and 0.24, Bluepill and Blackpill. Write still fails on 0.24 after an erase.

Erasing c=0-83:h=0-1

Erasing Track 0.0...
Erasing Track 0.1...
Erasing Track 1.0...
Erasing Track 1.1...
Erasing Track 2.0...
Erasing Track 2.1...
Erasing Track 3.0...
Erasing Track 3.1...
Erasing Track 4.0...
Erasing Track 4.1...
Erasing Track 5.0...
Erasing Track 5.1...
Erasing Track 6.0...
Erasing Track 6.1...
Erasing Track 7.0...
Erasing Track 7.1...
Erasing Track 8.0...
Erasing Track 8.1...
Erasing Track 9.0...
Erasing Track 9.1...
Erasing Track 10.0...
Erasing Track 10.1...
Erasing Track 11.0...
Erasing Track 11.1...
Erasing Track 12.0...
Erasing Track 12.1...
Erasing Track 13.0...
Erasing Track 13.1...
Erasing Track 14.0...
Erasing Track 14.1...
Erasing Track 15.0...
Erasing Track 15.1...
Erasing Track 16.0...
Erasing Track 16.1...
Erasing Track 17.0...
Erasing Track 17.1...
Erasing Track 18.0...
Erasing Track 18.1...
Erasing Track 19.0...
Erasing Track 19.1...
Erasing Track 20.0...
Erasing Track 20.1...
Erasing Track 21.0...
Erasing Track 21.1...
Erasing Track 22.0...
Erasing Track 22.1...
Erasing Track 23.0...
Erasing Track 23.1...
Erasing Track 24.0...
Erasing Track 24.1...
Erasing Track 25.0...
Erasing Track 25.1...
Erasing Track 26.0...
Erasing Track 26.1...
Erasing Track 27.0...
Erasing Track 27.1...
Erasing Track 28.0...
Erasing Track 28.1...
Erasing Track 29.0...
Erasing Track 29.1...
Erasing Track 30.0...
Erasing Track 30.1...
Erasing Track 31.0...
Erasing Track 31.1...
Erasing Track 32.0...
Erasing Track 32.1...
Erasing Track 33.0...
Erasing Track 33.1...
Erasing Track 34.0...
Erasing Track 34.1...
Erasing Track 35.0...
Erasing Track 35.1...
Erasing Track 36.0...
Erasing Track 36.1...
Erasing Track 37.0...
Erasing Track 37.1...
Erasing Track 38.0...
Erasing Track 38.1...
Erasing Track 39.0...
Erasing Track 39.1...
Erasing Track 40.0...
Erasing Track 40.1...
Erasing Track 41.0...
Erasing Track 41.1...
Erasing Track 42.0...
Erasing Track 42.1...
Erasing Track 43.0...
Erasing Track 43.1...
Erasing Track 44.0...
Erasing Track 44.1...
Erasing Track 45.0...
Erasing Track 45.1...
Erasing Track 46.0...
Erasing Track 46.1...
Erasing Track 47.0...
Erasing Track 47.1...
Erasing Track 48.0...
Erasing Track 48.1...
Erasing Track 49.0...
Erasing Track 49.1...
Erasing Track 50.0...
Erasing Track 50.1...
Erasing Track 51.0...
Erasing Track 51.1...
Erasing Track 52.0...
Erasing Track 52.1...
Erasing Track 53.0...
Erasing Track 53.1...
Erasing Track 54.0...
Erasing Track 54.1...
Erasing Track 55.0...
Erasing Track 55.1...
Erasing Track 56.0...
Erasing Track 56.1...
Erasing Track 57.0...
Erasing Track 57.1...
Erasing Track 58.0...
Erasing Track 58.1...
Erasing Track 59.0...
Erasing Track 59.1...
Erasing Track 60.0...
Erasing Track 60.1...
Erasing Track 61.0...
Erasing Track 61.1...
Erasing Track 62.0...
Erasing Track 62.1...
Erasing Track 63.0...
Erasing Track 63.1...
Erasing Track 64.0...
Erasing Track 64.1...
Erasing Track 65.0...
Erasing Track 65.1...
Erasing Track 66.0...
Erasing Track 66.1...
Erasing Track 67.0...
Erasing Track 67.1...
Erasing Track 68.0...
Erasing Track 68.1...
Erasing Track 69.0...
Erasing Track 69.1...
Erasing Track 70.0...
Erasing Track 70.1...
Erasing Track 71.0...
Erasing Track 71.1...
Erasing Track 72.0...
Erasing Track 72.1...
Erasing Track 73.0...
Erasing Track 73.1...
Erasing Track 74.0...
Erasing Track 74.1...
Erasing Track 75.0...
Erasing Track 75.1...
Erasing Track 76.0...
Erasing Track 76.1...
Erasing Track 77.0...
Erasing Track 77.1...
Erasing Track 78.0...
Erasing Track 78.1...
Erasing Track 79.0...
Erasing Track 79.1...
Erasing Track 80.0...
Erasing Track 80.1...
Erasing Track 81.0...
Erasing Track 81.1...
Erasing Track 82.0...
Erasing Track 82.1...
Erasing Track 83.0...
Erasing Track 83.1...

 erase --tracks="c=0-83:h=0-1:step=1"

keirf, 2021-01-23:
Could you attach a photo of your setup?

M1kerochip, 2021-01-23:
Sure:
https://i.imgur.com/KQXN08y.jpg

and F1s
https://i.imgur.com/I9BIiJ4.jpg

If you want better photos, let me know.

I made a floppy female to female  power lead, and just swapped it with duponts, just in case.

One thing I have noticed though:
If you recall, earlier, I said the gw.exe update wasn't working for me:
up to v0.20 the update command and .upd file work fine, and from 0.22-0.24, that doesn't work.
I have to manually flash the .hex to upgrade to those versions.

Is it something to do with the version of Python bundled?? Has that changed?

Can I install python manually, the required libraries, and then run the python scripts under windows still?

M1kerochip, 2021-01-23:
They're not amazing photos, I guess. I can get something better, if you like. (The white piece of tape on the APM32 is to dampen the lights)

I've tried different usb calbes too, including a mini cable, and a standard 1.8m one (Both work fine with other usb devices)

keirf, 2021-01-23:
Okay the update problem is strange and unusual too. What exactly happens. You should *always* be able to `gw update` with any tools version, and the firmware will be updated to the version matching the tools.

M1kerochip, 2021-01-23:
Nothing.

If I run either gw.exe update, or gw.exe update <path_to_.upd> it just hangs.

I've left it for a few mins, and the screen stays sitting, waiting, nothing extra is output.

If I end task the gw.exe, I get an error about needing some buffers:

** FATAL ERROR:
unpack requires a buffer of 2 bytes

But, if not, it just sits there forever.

keirf, 2021-01-23:
Can you `gw --bt update` and bail out of the hang, and report the back trace you should see printed out.

M1kerochip, 2021-01-23:
Doesn't seem to do anything:

F:\gw\Greaseweazle-v0.24>gw --bt update
Updating Main Firmware to v0.24...

F:\gw\Greaseweazle-v0.24>

Takes a long time to end the gw.exe task, too. (where usually it's within a second)

(Same from cmd.exe or Windows Terminal)

keirf, 2021-01-24:
Okay well it is interesting that v0.22+ is precisely what fails for you in two ways: Writes and Updates.

If you run an old bootloader, by flashing with HEX to say v0.16, can you update the main firmware to v0.24 using the `gw update` method? It is weird if not as neither the bootloader and update script have not really changed since v0.21 or earlier (and of course you'd be running a bootloader [v0.16] that you know works!).

You can indeed run the Python scripts in Windows, just install Python 3 and the required libraries. You may get a warning about running "setup.sh" for accelerated routines, but you can ignore that for testing.

M1kerochip, 2021-01-24:
> If you run an old bootloader, by flashing with HEX to say v0.16, can you update the main firmware to v0.24 using the `gw update` method? It is weird if not as neither the bootloader and update [...]

yes, that's exactly what I did, but, I did it with the 0.17 gw.exe version.

> You can indeed run the Python scripts in Windows, just install Python 3 and the required libraries. You may get a warning about running "setup.sh" for accelerated routines, but you can ignore that for testing.

Same. 0.16 firmware, 0.16 tools, 0.24 update, Python 3.8.1 (which is what I used before) I get:

D:\Programming\Python38>python F:\gw\Software\Greaseweazle\Greaseweazle-v0.16\gw.py update F:\gw\Software\Greaseweazle\Greaseweazle-v0.24\Greaseweazle-v0.24.upd

** Bootloader v0.16 [F1], Host Tools v0.16
** FATAL ERROR:
unpack requires a buffer of 2 bytes

Device remains in update mode, no crashing, no lockup, immediately displays the error, and returns to command line.
 
> EDIT: Also did you try another host (your laptop) yet?
Yes, flashed 0.24 .hex, and tried. Exactly the same. Failed to write 0.0 - write protected.

keirf, 2021-01-24:
You can't update to later firmware from earlier tools. In fact the later tools are explicitly locked to only update to exact version-matched firmware. So, to update to v0.24 firmware, run `gw update` from v0.24 tools.

I assume you are running the v0.24 tools with v0.24 firmware? I think you have to, actually, at least I did version-lock tools to firmware from the very beginning...

M1kerochip, 2021-01-24:
OK, update using the 0.16 firmware, with 0.24 tools, and 0.24 .upd works fine.

D:\Programming\Python39>python F:\gw\Software\Greaseweazle\Greaseweazle-v0.24\gw.py update F:\gw\Software\Greaseweazle\Greaseweazle-v0.24\Greaseweazle-v0.24.upd
*** WARNING: Optimised data routines not found: Run scripts/setup.sh
Updating Main Firmware to v0.24...
Done.
** Disconnect Greaseweazle and remove the Programming Jumper.

OK, so, 0.16 firmware, 0.24 tools and host:
D:\Programming\Python39>python F:\gw\Software\Greaseweazle\Greaseweazle-v0.24\gw.py write D:\Programming\Projects\sdbox-master\IDE-Setup.adf
*** WARNING: Optimised data routines not found: Run scripts/setup.sh
Writing c=0-81:h=0-1
Writing Track 1.1...Command Failed: WriteFlux: Disk is Write Protected

It tries to write 0.0 and 0.1 and 1.0 and fails on 1.1

That's the first time a 0.24 gw didn't fail on 0.0

keirf, 2021-01-24:
Please can you grab a backtrace: `--bt` after `gw.py` on command line.

M1kerochip, 2021-01-24:
```
D:\Programming\Python39>python F:\gw\Software\Greaseweazle\Greaseweazle-v0.24\gw.py --bt write D:\Programming\Projects\sdbox-master\IDE-Setup.adf
*** WARNING: Optimised data routines not found: Run scripts/setup.sh
Writing c=0-81:h=0-1
Writing Track 2.0...Command Failed: WriteFlux: Disk is Write Protected
```

Nothing after that.

keirf, 2021-01-24:
Oh I swallow the error. That's annoying. Can you put a `raise` on line 93 of scripts/greaseweazle/tools/write.py? It would go aligned with and directly underneath the print() statement. I don't know if you're handy with Python at all. If not I can roll you a build to try...

M1kerochip, 2021-01-24:
I've never used python!
But putting raise on that line gives me:

TabError: inconsistent use of tabs and spaces in indentation

keirf, 2021-01-24:
Ah no worries. Let me make a branch and I can send you test versions to try out.

keirf, 2021-01-24:
Okay, can you download the artifact from this build run: https://github.com/keirf/Greaseweazle/actions/runs/507423735
1. You'll need to be logged in to github
2. Link is then at the bottom of the page
3. The download is a zip containing three zips. You want the inner zip without `-win` or `-debug` in the name. That's a plain Python-source distribution.
4. Unzip it, `gw update` to the firmware within, and try your gw write command. No need for `--bt` as that's hardcoded here.

Let me know what backtrace you get :)

M1kerochip, 2021-01-24:
D:\Programming\Python39>python F:\gw\Software\Greaseweazle\Greaseweazle-fa0cd7d\gw.py write D:\Programming\Projects\sdbox-master\IDE-Setup.adf

gives 

*** TEST/PRE-RELEASE: commit fa0cd7d508913f3cb939e3db0e9db83e05f0558e
*** Use these tools and firmware ONLY for test and development!!
*** WARNING: Optimised data routines not found: Run scripts/setup.sh
Writing c=0-81:h=0-1
Writing Track 1.1...Command Failed: WriteFlux: Unknown Error (20)
Traceback (most recent call last):
  File "F:\gw\Software\Greaseweazle\Greaseweazle-fa0cd7d\gw.py", line 11, in <module>
    import gw
  File "F:\gw\Software\Greaseweazle\Greaseweazle-fa0cd7d\scripts\gw.py", line 75, in <module>
    res = main(argv)
  File "F:\gw\Software\Greaseweazle\Greaseweazle-fa0cd7d\scripts\greaseweazle\tools\write.py", line 184, in main
    util.with_drive_selected(write_from_image, usb, args, image)
  File "F:\gw\Software\Greaseweazle\Greaseweazle-fa0cd7d\scripts\greaseweazle\tools\util.py", line 184, in with_drive_selected
    fn(usb, args, *_args, **_kwargs)
  File "F:\gw\Software\Greaseweazle\Greaseweazle-fa0cd7d\scripts\greaseweazle\tools\write.py", line 84, in write_from_image
    usb.write_track(flux_list = flux_list,
  File "F:\gw\Software\Greaseweazle\Greaseweazle-fa0cd7d\scripts\greaseweazle\usb.py", line 431, in write_track
    raise error
  File "F:\gw\Software\Greaseweazle\Greaseweazle-fa0cd7d\scripts\greaseweazle\usb.py", line 420, in write_track
    self._send_cmd(struct.pack("4B", Cmd.WriteFlux, 4,
  File "F:\gw\Software\Greaseweazle\Greaseweazle-fa0cd7d\scripts\greaseweazle\usb.py", line 205, in _send_cmd
    raise CmdError(cmd, r)
greaseweazle.usb.CmdError: WriteFlux: Unknown Error (20)

keirf, 2021-01-24:
Hmm well, a write command was sent to the firmware, and it really did detect that the write-protect line (floppy pin 28; BluePill pin B8) is asserted. I changed the error code to that one place in the firmware to 20, and that is being reported through the tools.

You say it's on multiple drives and multiple disks and multiple ribbon cables, which ought to discount the possibility of a dodgy switch, disk or cable. Maybe we should take a closer look at your adapter(s)?

This does smell like a floating input pin getting randomly pulled low.

keirf, 2021-01-24:
Can you do a real close up of your adapter, especially where the resistors attach. Did you solder them up yourself?

M1kerochip, 2021-01-24:
Yeah, I soldered them myself.

I tried connecting one of the other drives, just with dupont, and ... sort of success.

It writes until track 75.1 and fails. Consistently. (So maybe it's a disk error? This drive isn't as good as the Amiga one. The disk did write fine on the Teac drive though)

D:\Programming\Python39>python F:\gw\Software\Greaseweazle\Greaseweazle-fa0cd7d\gw.py write D:\Programming\Projects\sdbox-master\IDE-Setup.adf
*** TEST/PRE-RELEASE: commit fa0cd7d508913f3cb939e3db0e9db83e05f0558e
*** Use these tools and firmware ONLY for test and development!!
*** WARNING: Optimised data routines not found: Run scripts/setup.sh
Writing c=0-81:h=0-1
Writing Track 75.1...Traceback (most recent call last):
  File "F:\gw\Software\Greaseweazle\Greaseweazle-fa0cd7d\gw.py", line 11, in <module>
    import gw
  File "F:\gw\Software\Greaseweazle\Greaseweazle-fa0cd7d\scripts\gw.py", line 75, in <module>
    res = main(argv)
  File "F:\gw\Software\Greaseweazle\Greaseweazle-fa0cd7d\scripts\greaseweazle\tools\write.py", line 184, in main
    util.with_drive_selected(write_from_image, usb, args, image)
  File "F:\gw\Software\Greaseweazle\Greaseweazle-fa0cd7d\scripts\greaseweazle\tools\util.py", line 184, in with_drive_selected
    fn(usb, args, *_args, **_kwargs)
  File "F:\gw\Software\Greaseweazle\Greaseweazle-fa0cd7d\scripts\greaseweazle\tools\write.py", line 107, in write_from_image
    error.check(verified, "Failed to write Track %u.%u" % (cyl, head))
  File "F:\gw\Software\Greaseweazle\Greaseweazle-fa0cd7d\scripts\greaseweazle\error.py", line 15, in check
    raise Fatal(desc)
greaseweazle.error.Fatal: Failed to write Track 75.1

Trying to read in that disk gives me a memory error:

D:\Programming\Python39>python F:\gw\Software\Greaseweazle\Greaseweazle-fa0cd7d\gw.py read test.scp
*** TEST/PRE-RELEASE: commit fa0cd7d508913f3cb939e3db0e9db83e05f0558e
*** Use these tools and firmware ONLY for test and development!!
*** WARNING: Optimised data routines not found: Run scripts/setup.sh
Reading c=0-81:h=0-1 revs=3
T0.0: Raw Flux (147429 flux in 690.79ms)
T0.1: Raw Flux (163035 flux in 756.21ms)
T1.0: Raw Flux (177459 flux in 741.84ms)
T1.1: Raw Flux (151933 flux in 728.32ms)
T2.0: Raw Flux (154368 flux in 752.29ms)
T2.1: Raw Flux (157374 flux in 755.70ms)
T3.0: Raw Flux (157041 flux in 747.87ms)
T3.1: Raw Flux (164248 flux in 753.56ms)
T4.0: Raw Flux (161323 flux in 733.98ms)
T4.1: Raw Flux (161687 flux in 741.46ms)
T5.0: Raw Flux (156189 flux in 742.14ms)
T5.1: Raw Flux (158283 flux in 753.78ms)
T6.0: Raw Flux (157175 flux in 746.63ms)
T6.1: Raw Flux (157053 flux in 754.08ms)
T7.0: Raw Flux (155891 flux in 742.62ms)
T7.1: Raw Flux (158227 flux in 756.48ms)
T8.0: Raw Flux (159030 flux in 748.92ms)
T8.1: Raw Flux (162309 flux in 750.94ms)
T9.0: Raw Flux (161417 flux in 742.02ms)
T9.1: Raw Flux (162837 flux in 743.32ms)
T10.0: Raw Flux (153123 flux in 741.18ms)
T10.1: Raw Flux (169615 flux in 760.61ms)
T11.0: Raw Flux (162582 flux in 734.88ms)
T11.1: Raw Flux (157596 flux in 747.99ms)
T12.0: Raw Flux (161095 flux in 748.03ms)
T12.1: Raw Flux (160098 flux in 744.54ms)
T13.0: Raw Flux (162179 flux in 741.74ms)
T13.1: Raw Flux (165063 flux in 747.29ms)
T14.0: Raw Flux (161731 flux in 738.56ms)
T14.1: Raw Flux (163276 flux in 744.89ms)
T15.0: Raw Flux (155508 flux in 739.79ms)
T15.1: Raw Flux (150365 flux in 755.04ms)
T16.0: Raw Flux (159298 flux in 760.12ms)
T16.1: Raw Flux (153773 flux in 751.76ms)
T17.0: Raw Flux (157728 flux in 754.78ms)
T17.1: Raw Flux (149359 flux in 727.13ms)
T18.0: Raw Flux (162770 flux in 754.88ms)
T18.1: Raw Flux (150790 flux in 746.36ms)
T19.0: Raw Flux (157389 flux in 754.10ms)
T19.1: Raw Flux (157373 flux in 754.37ms)
T20.0: Raw Flux (156035 flux in 747.60ms)
T20.1: Raw Flux (161697 flux in 753.80ms)
T21.0: Raw Flux (174628 flux in 740.98ms)
T21.1: Raw Flux (157029 flux in 729.13ms)
T22.0: Raw Flux (168620 flux in 741.56ms)
T22.1: Raw Flux (160893 flux in 733.39ms)
T23.0: Raw Flux (169255 flux in 739.47ms)
T23.1: Raw Flux (159216 flux in 729.42ms)
T24.0: Raw Flux (161246 flux in 746.14ms)
T24.1: Raw Flux (164413 flux in 746.07ms)
T25.0: Raw Flux (155534 flux in 740.45ms)
T25.1: Raw Flux (159833 flux in 754.85ms)
T26.0: Raw Flux (160711 flux in 744.75ms)
T26.1: Raw Flux (161994 flux in 748.61ms)
T27.0: Raw Flux (157138 flux in 741.03ms)
T27.1: Raw Flux (158975 flux in 753.81ms)
T28.0: Raw Flux (156902 flux in 748.36ms)
T28.1: Raw Flux (159405 flux in 754.82ms)
T29.0: Raw Flux (156599 flux in 744.79ms)
T29.1: Raw Flux (158322 flux in 752.41ms)
T30.0: Raw Flux (157687 flux in 747.78ms)
T30.1: Raw Flux (155724 flux in 751.86ms)
T31.0: Raw Flux (160747 flux in 752.13ms)
Traceback (most recent call last):
  File "F:\gw\Software\Greaseweazle\Greaseweazle-fa0cd7d\gw.py", line 11, in <module>
    import gw
  File "F:\gw\Software\Greaseweazle\Greaseweazle-fa0cd7d\scripts\gw.py", line 75, in <module>
    res = main(argv)
  File "F:\gw\Software\Greaseweazle\Greaseweazle-fa0cd7d\scripts\greaseweazle\tools\read.py", line 155, in main
    util.with_drive_selected(read_to_image, usb, args, image,
  File "F:\gw\Software\Greaseweazle\Greaseweazle-fa0cd7d\scripts\greaseweazle\tools\util.py", line 184, in with_drive_selected
    fn(usb, args, *_args, **_kwargs)
  File "F:\gw\Software\Greaseweazle\Greaseweazle-fa0cd7d\scripts\greaseweazle\tools\read.py", line 105, in read_to_image
    image.emit_track(cyl, head, dat)
  File "F:\gw\Software\Greaseweazle\Greaseweazle-fa0cd7d\scripts\greaseweazle\image\scp.py", line 184, in emit_track
    flux.cue_at_index()
  File "F:\gw\Software\Greaseweazle\Greaseweazle-fa0cd7d\scripts\greaseweazle\flux.py", line 49, in cue_at_index
    self.list = [-to_index] + self.list[i+1:]
MemoryError

On the PCB:
Close: Not great photos, the camera doesn't focus when I'm too close.
https://i.imgur.com/0mG4Ehs.jpg
https://i.imgur.com/87g90Yp.jpg
https://i.imgur.com/4VYvRg6.jpg
https://i.imgur.com/D5ONznV.jpg

M1kerochip, 2021-01-24:
OK, I think you can close this.

Putting the F1 that was connected via the duponts back in the PCB, it immediately gives a "Disk is Write Protected" error, where it wrote to track 75 earlier, with the duponts (but the Epson drive)

But. I flashed it back to 0.20. And it wrote the disk no problem.
So. Whatever the reason, 0.22+ is failing on the write, where 0.20 isn't.

I guess, I'll remove the resistors, and clean the PCB, and put new ones on.
They're 4x 0805 1K 1% resistors. (I didn't measure the resistance before I installed them, I just assumed they were fine)

Just touching one side of the solder to the other with the multimeter, it shows a reading of 996 ohms on all four resistors, but, I'm not sure which pins they connect to on the PCB to test properly.

I have more PCBs but I've misplaced them! Otherwise I'd just try another one.

keirf, 2021-01-24:
Regarding your adapter, those solder joints are not good. Can you see on some of the resistors, on one side, the solder has not spread onto the PCB pad? That is not a proper electrical connection: It will look connected some of the time, but depending how the PCB flexes or heats, the junction will show high resistance. You do not need a new adapter: You need to reflow those joints with flux and touch an iron to them. The flux will cause the solder to flow nicely.

To your other errors:
1."Failed to write Track 75.1": This is a media failure where that ADF track has failed to verify after write, after three attempts. The error really needs to be clarified (eg Verify error on Track 75.1) and I will be doing that for next release.
2. Haven't seen MemoryError before. Yes there will be possibly 100s of MBs held in memory before writing the image file. But modern systems should have gigabytes so not a problem. :) Is your host system here memory constrained?

keirf, 2021-01-24:
Regarding the bad electrical connection, I suspect that the resistor for floppy pin 28 doesn't connect at all, and that line floats when not asserted by the drive. In this case the line can read any value at any time. It is pure chance that it works with earlier firmwares and less well with later ones.

M1kerochip, 2021-01-24:
AH HA!

OK.

After writing down all my testing:
I finally found the problem(s).

It's this board, and the resistor(s).
https://robotdyn.com/black-pill-apm32f103cb-128kb-flash-20kb-sram-stm32-compatible-arm-cortexr-m3-mcu-mini-board.html

That particular "blackpill" has an APM32F103 onboard. And runs at 96Mhz. If that makes a difference.

I never noticed, since it's almost identical to the other STM32F103 I have from RobotDYN.

It reads perfectly, but, since 0.22+ it won't write.

Resoldering the four resistors:
Now each of the STM32F103 boards reads and writes correctly, either on duponts, or on the PCB.

However!
The APM32F103 fails, randomly, to write on the PCB now. It sometimes fails on track 0.1 but mostly on 0.0.
It fails on higher track numbers only connected via duponts, but it *always* fails.
Reflashing 0.20 on it, and it writes ok, on the PCB or direct connect.

And of course, the first board I tried, connected via the duponts, was the APM32 board.
And connected the STM32 board on firmware 0.20, so.
It's only making notes, and testing properly, that I found it.

SO. Mystery solved, more or less.

The MemoryError: I've been able to recreate that, too.
I opened a webpage and it ate up huge amount of ram, crashed, didn't release the ram, and continued to chew ram.
I never noticed until later. When I eventually went back through my steps, yes, I'd run out of memory, I'm guessing.
(I killed chrome at one stage, and it was after the memory error, so I'm guessing that's the cause)
(Don't browse and use Chrome kids!)

If I use enough ram, I can get the memory error. (I have to open a hundred+ web pages, but it happens eventually.)

I can post you one of those APM32 boards, if you like. I think I have one or two of them here unused. They're about €2, which is why I bought some on my last order, to test them out. They worked fine, so I left them in the PCB.

keirf, 2021-01-24:
Yes I'd be happy to take a look at one. My email is keir.xen@gmail.com so drop me a message to get my mailing address. Cheers!

keirf, 2021-01-24:
So to be sure: The APM32F103 *still* is failing specifically with the "Disk is Write Protected" error?

M1kerochip, 2021-01-24:
Yeah. The AMP32, and only that board, is failing with that specific error.
It also fails to update to 0.22+, with the .upd file.
(So, just the bare board connected via the USB to the PC)

B8 looks fine, I think.

This particular APM32 I bought already soldered, and came with a QA sticker on it.

I can open another one, solder on the legs, and see if the new one is the same?

keirf, 2021-01-24:
Yes I would. And if so you still have one to send me.

M1kerochip, 2021-01-24:
OK! Yeah, can do.
A job for tomorrow though, I think.

M1kerochip, 2021-01-25:
OK! Final update:

I soldered up a new APM32, and it worked fine on reading/writing! So! Good news there.
It still, however, fails to run the .upd file.

I reflowed the solder on B8 on the pre-soldered one i bought: And it now successfully writes properly too!
(But still fails on the update)
SO!

Good news, I guess. I still have one APM32, which I can post.

Photo of the STM32F103 on the right, and APM32F103 on the left. https://i.imgur.com/Szl780E.jpg
(Easy to mix up!) the STM has a proper indent on the chip. https://i.imgur.com/besftHM.jpg

The soldered/non soldered APM32 packages: https://i.imgur.com/okisHi2.jpg

Somewhere along the way, my proper STM32F103 RobotDYN board has died.
The PC no longer picks it up.

In conclusion: All good, and I'll post the APM towards the end of the week :)

keirf, 2021-01-25:
So you say it *used* to run the update but now doesn't? Or has it never worked on APM32?

You jumper SCK-GND, and the LED should flash and you should be in bootloader mode. But that doesn't happen?

M1kerochip, 2021-01-25:
Yup.

The APM32 ran the .upd file up to 0.20, and on 0.21+ it just hangs instead of flashing.

So, the initial problem I had can be broken down in to 3 problems:

1) I can't update the APM32F103 using the .upd file on 0.21+

2) APM32 failed to write connected via duponts,

3) APM32 failed to write connected via the PCB.
~
2) was solved reflowing the solder on B8.

3) was solved resoldering the resistors on the PCB.

Issue 1) still remains.

keirf, 2021-01-25:
Ok, but you can update via HEX file on the APM32 still?

It sounds like the APM32 crashes, if the LED goes out... Maybe I can tweak the update process to still work with those chips.

keirf, 2021-01-25:
I can safely say nothing has changed between v0.20 and v0.21 which should affect the update bootloader in any way. Except the main firmware has got fractionally bigger. The F1 update payload in v0.20 is 9660 bytes. In v0.21 it is 9760 bytes.

Do you know what version bootloader you have in place when you do these tests (you can also check that with `gw info` while in bootloader mode).

M1kerochip, 2021-01-25:
I can update via the .hex file fine, each time.

I flashed the .hex 0.16 - 0.20 yesterday, and it updated to 0.24 ok.
Flashing the 0.24 .hex today, and it won't update to 0.24.

I can try 0.22 and 0.23 hex files too?

keirf, 2021-01-25:
Okay, so with up to v0.20 HEX (and therefore v0.20 bootloader) you could update to v0.24 okay via `gw update`.

But with v0.24 bootloader you cannot "update" to v0.24.

I suppose v0.22, say, could be interesting. Or v0.21 if you believe that bootloader doesn't work either.

keirf, 2021-02-01:
Hi, Just wanted to check if you had a chance to post the APM32? There are other reports of the update problem. I should really fix it or blacklist the chip and detect it in my Blinky Test.

closed 2021-02-15 by keirf

keirf, 2021-03-16:
Received the APM32F103 today, thank you!

However, my tests with v0.25 have not raised any issues with `gw update`. So far...

keirf, 2021-03-16:
Tried flashing from v0.19 to v0.24, v0.22 to v0.24, etc etc. No hangs, no issues.

keirf, 2021-03-16:
I'll hang on to this APM32 in case of issue reports in future though!

keirf, 2021-03-16:
A picture of my APM32 chip

[image]

keirf, 2021-03-16:
I will take a guess that 2023 is a date code, 23rd week of 2020.

[3 more replies]