keirf/greaseweazle #76: Feature request : Test input pin

ajefr, 2021-02-14:
Hello,
I have some trouble with a 3" drive and cannot figure if it comes from the drive or from the greaseweazle.
I tested two greaseweazle f7 lightning plus and both have the same problem.
On the 3" drive, read head did not come back to track 0. The drive head move to home but sometime move only 1-2mm, sometimes more but never at the exact home position
With an oscilloscope, I can see that the sensor is working and also the signal on the connector going down when the head reach the home.
May it be possible to have a input test feature (like the output) to be sure that it did not comes from the STM32 ?
Thanks

keirf, 2021-02-15:
Yes, though in this case, if the seek does not fail with NoTrk0 ("Track 0 not found") then Greaseweazle is seeing the TRK0 pin go low. I suppose it could be thinking it goes low when it hasn't?

keirf, 2021-02-15:
Also you can try decreasing the seek speed by increasing time between step pulses. For example:
```
gw delays --step 10000
````

Will double the time between steps to 10ms. Perhaps your drive requires this?

ajefr, 2021-02-19:
Hi
Sorry for late reply.
So I tried the time increase and the head looks to move better now.
I'm not absolutely sure that it ends exactly at the sensor position but it's far better than previous.
Is the delay parameter is saved on RAM or on EEPROM ?
As I have dedicaced one board per drive, it will be great if it's definitively saved on board in case I forgot parameter in the future.

keirf, 2021-02-21:
Delay parameters are not currently saved. There's no handy small Flash area to use on F7 boards as the Flash sector size is 16kB which is a quarter of the total. EEPROM bytes would be too little to save the entire delays parameter block. Pushing the parameters down from the host would be a nice model if there was a standard GUI everyone used. It's not so great for current command-line usage.

ourIThome, 2021-02-21:
Ajefr you might as a work around be able to alter gw file, so it's hard coded with the parameters, which then calls the main gw python file.  Then you would call your custom slow python file, when you know you need to use the slower parameters.  I could help and take a look for you, if your not using the pre bundled python / gw.exe file.

keirf, 2021-02-21:
There's been discussion of having an INI file (#45) which could be read on every command-line invocation.

Alternatively, on Linux it could also be done in udev rules on device hotplug. But I don't know if there's an equivalent hook that can be triggered in Windows.

closed 2021-12-31 by keirf