keirf/greaseweazle #77: Wiki request: add list of error code blinks to the Troubleshooting FAQ

philpem, 2021-02-18:
I built a Greaseweazle F7 Lightning Plus and had an issue with bad soldering on XO1 (CPU clock oscillator). This led to the oscillator not starting up, and the board failing to appear in `lsusb`.

The board flashed the ACT LED three times to indicate the oscillator failure -- which lacking documentation,  I misinterpreted as "waiting for USB" and thus bad soldering on the USB connector. Had I known it was an oscillator fault, my debugging process could have been shortened :)

From the code (added in 3276cf7c388226e75cd306db63abd4ea22b9bb8d) it looks like the currently-valid error blinks are:

  * 1 blink: board version straps somehow indicated high and low (check for bad soldering around those pins)
  * 2 blinks: board version straps not recognised (check you're loading the correct firmware)
  * 3 blinks: oscillator failed to start up (check clock oscillator)

Could these be added to the Troubleshooting FAQ on the Wiki?

Another point of note is that it's easy to accidentally flood the clock oscillator with solder so that solder flows up the plated contact channels on the side, and shorts against the can. This can be hard to spot visually. I found removing the solder with wick and reapplying a smaller amount was a good way to fix this. Solder only needs to form a fillet at the bottom of the channel to make contact with the XO.

keirf, 2021-02-18:
Yes I'll add this. I added the 3 blinks specifically for this situation as it's so easy to mess up the XO. I never managed to recover one before so it's good to know it can be done!

keirf, 2021-02-18:
I have added text to the Troubleshooting FAQ, and I have also added an extra tip to the F7L+ assembly guide. See what you think.

closed 2021-02-18 by keirf