keirf/greaseweazle #78: Feature request: DC Erase, Write Test Tracks

philpem, 2021-02-18:
It would be useful for Greaseweazle to be able to DC-erase all or a subset of the tracks on a disc. This is done by enabling the write gate but sending no flux transitions.

This can be useful when writing images to be read in 40-track (48tpi) drives, when the writing drive has an 80-track (96tpi) head. Every other track is DC-erased to prevent them from interfering with the data on the 'real' track.

This could be even more useful if Greaseweazle were able to write some basic test patterns to the disc:

  * Index burst (burst of transitions 200us after the index). Used to align the Index sensor.
  * 1F and 2F test patterns (a track of '10' and '11' FM clock-data sequences at a given data rate). Used to check head performance, and can be used to align the Track Zero sensor or do a coarse head alignment.
  * 1F+2F mixed - bursts of 1F followed by bursts of 2F, repeated for the whole track. Used to check head and read-channel performance.

With these patterns, a basic (functional) alignment can be done on another drive -- provided the user has a known well-aligned drive.

If Greaseweazle had "step in" and "step out" commands (as opposed to 'seek to specific cylinder'), it would also be possible to do Track Zero sensor alignment.

keirf, 2021-02-18:
Greaseweazle has `gw erase` command for DC-erase. It's not as good as a strong magnet though (eg neodymium)

We could make a test disk IPF?

philpem, 2021-02-18:
So it does! Sorry, I don't know how I missed that. As you say, nowhere near as good as a nice strong magnet, but good enough in a pinch.

A test-disk IPF would be good - I was toying with doing that with an SCP file, but haven't had a good look at the SCP spec to see if that's possible. Configurability would be even better - maybe have a Python script which takes a list of tracks and contents and produces an SCP or IPF file.

It would be good to have "seek up" (away from zero) and "seek down" (toward zero) commands, especially if they included feedback about whether track zero had been hit. I'm not sure how feasible this is - I haven't looked at the Greaseweazle code yet. Just throwing ideas around!

keirf, 2021-02-18:
Yes Python script's a good idea.

For seek up/down, what's the usage? Seek(0) is always done out to TRK0 sensor.

philpem, 2021-02-18:
The alignment process for the Teac FD-55 is on page 154 of http://www.bitsavers.org/pdf/teac/10180320_FD55_Maint_1985.pdf

You'd use an alignment disc (e.g. Dysan 224/2A for a 5.25in drive, spec: http://www.bitsavers.org/pdf/dysan/224-2a_5inch_DS_AlignmentDiscSpec_Aug79.pdf) and a blank work disc.

* Move the track zero sensor back slightly
* With the alignment disc inserted, spin up and manually seek to the alignment track (track 16). Check the drive alignment. Stop the motor.
* Remove the Alignment disc and insert the Work disc. Start the motor.
* Seek back 15 tracks (now at track 1) and adjust the sensor so that it's just off.
* Seek up 2 tracks and back 3. Check the sensor clearly indicates track zero.

It's a bit of a faff, and I'm sure there are many other ways to achieve the same goal. Either way, you need relative positioning (up/down or +n/-n) without using the track-zero sensor.

The same relative-seek feature is useful when aligning drives - because you have to seek in until you find the alignment track on track 16 or 32. If the drive alignment is completely untrustworthy (ie. the drive has been played with by a "screwdriver expert") then you can't rely on track-zero or the track centreline alignment.

keirf, 2021-02-18:
Ok, I can add a relative track seek which also returns the trk0 sensor value after the seek.

keirf, 2021-02-18:
That plus an IPF test disk could do the job. Possibly plus an alignment-test command could be useful to run through the steps?

philpem, 2021-02-18:
Sounds good to me!

keirf, 2021-02-21:
@philpem Do you know what the index burst track looks like? eg. 125kHz except no flux for the burst duration? Or vice versa? Or something else?

philpem, 2021-02-21:
I think the Dysan spec or possibly the Dymek manual have an example. It's a burst of 1F transitions 200 microseconds after the index pulse, then the rest of the track is blank.

keirf, 2021-02-21:
Blank means the drive returns random/unformatted flux though. Should it look like a No Flux Area: That is, no flux transitions on read? As that is mastered by rapid approx 1MHz flux transitions on write.

philpem, 2021-02-21:
No flux transitions is probably best for ease of use.

keirf, 2021-02-21:
Ok, I'd probably plan to script this as an HFE image. Or write directly from a new greaseweazle (with some write verification). So that's easy either way.