keirf/greaseweazle #80: Graseweazle SCP format

merlinkv, 2021-02-18:
Hello Keirf.

I'm owner of one SuperCard Pro that uses the original SCP format to read/write disk and create iages, but the SCP files created with Greaseweazle are incompatible with the SuperCard Pro.

Why?

Any help will be much appreciated.

keirf, 2021-02-18:
Incompatible how? Greaseweazle version? They should be compatible.

XP is not supported by greaseweazle tools. I recommend at least Win7.

merlinkv, 2021-02-18:
SuperCard Pro can't read SCP files created with Greasewaezle.

SCPro software all times recognizes SCP files (Greaseweazle) as CP/M files.

merlinkv, 2021-02-18:
I have tested two SCP files created by a friend with Greasewaezle F1 (I think). As my friend said, these images transfered again to real disks works fine.

But I can't use these images with SuperCard Pro.

keirf, 2021-02-18:
Interesting. What do you want to do with them in supercard Pro? Does it matter they identify as cpm?

merlinkv, 2021-02-18:
Well, I want to write these images to real disks (Amiga), but I can't do it because are identidied as CP/M files.

keirf, 2021-02-18:
As with a lot of third party software, Greaseweazle doesn't bother setting the id header byte (that's the fourth byte). Supercard Pro seems to care about that byte a lot. If you know how you could try changing the fourth byte from hex value 80 to value 04. Value 04 means Amiga.

merlinkv, 2021-02-18:
Yes, that's the problem. I have changed the fourth byte to 04 make an now I have an usable image with SuperCard Pro.

It's possible to change this in the Greasewazle firm/soft? or is a problem?

Anyway, thanks you very much.

merlinkv, 2021-02-18:
Updated images (fourth byte to 04), can be usable with SCPro & WinUAE and can be written to real disk without issues.

keirf, 2021-02-18:
I will add a syntax to allow the SCP ID byte to be specified. Something like `gw read my.scp::disktype=4` I expect.

merlinkv, 2021-02-18:
Great! Thanks you very much.

commit 77131170 (2021-02-19): scp: New SCP option for gw read: 'disktype=<nr>|amiga|c64' sets ID byte
in SCP image header. Needed for correct interaction with original
Supercard Pro software (no other software seems to care about it).

For example: gw read a.scp::disktype=amiga

Fixes #80
closed 2021-02-19 by keirf