keirf/greaseweazle #81: Greaseweasle F1 0.24 5"1/4 disk drive : no index only when writing

Deksor, 2021-02-22:
Hello, I have a problem with my greaseweasle F1. I hooked everything up to a floppy disk cable (using dupont wires). I can read and I can write to 3.5" disks without any problems.
I can also read from 5"1/4 drives.
However when I try to write to 5"1/4 360K disks using two of my 5"1/4 DD drives and one of my 1.2MB drive, I get the error "Command Failed: GetFluxStatus: No Index"
Using another 5"1/4 1.2MB drive it works, but I have no idea what's the problem. Is my dupont wire thing too flimsy ? (The problem is systematic). Are my drives a bit broken (They were all tested good in PCs) ? Is it a problem with the F1 model ?

Also I get no proper images from one of the two 5"1/4 DD drives even though it works perfectly in my 8088.
Can you help ?

keirf, 2021-02-22:
You probably need pull up resistors on the drive outputs (inputs to Greaseweazle/blue pill). You could solder them to the blue pill or use an adapter pcb.

Deksor, 2021-02-22:
Do you mean the 4x 1K resistors that need to be put on the back of the PCB ?

I'll try and report back when I get my PCBs :)

keirf, 2021-02-22:
Yes those resistors.

Deksor, 2021-03-17:
I have made the PCB. It made major improvements. 
However the drive from my 8088 which works just refuses to work with the greaseweazle. I keep getting the No Index error.
Other drives that also work supposedly fine are having issues. When I open the .SCP files made with them in HxC floppy emulator's disk viewer, I can see many red circles. The drives are clean and they work in computers.
Is the F1 card more sensitive than PC's controllers ? Does that mean that these drives are actually bad ? Poorly calibrated ?

keirf, 2021-03-23:
Some drives need density select on pin 2. Try this before read/write: `gw pin 2 L`

Deksor, 2021-03-27:
Thanks. I tried it but it doesn't work. I mean the command works but the problem remains the same.
I could even check that the command worked using my 5"1/4 drive that can read and write both kind of floppy disks that outputs a different color on its activity led regarding the type of the disk (green = HD, amber = DD)

The images are still way off. All drives are known good, they work in computers using the exact same disks but the GW F1 (I even tried other GW F1) has seemingly problems using them. Do you want the images I made using them for analysis ? I can even take picture of the drives if you want (I tried some maintenance with them, cleaning the heads, lubricating the mechanisms ... without any change).

keirf, 2021-03-27:
Not especially. I would make a guess you need output buffers on your Greaseweazle. Some 5.25 drives have strong 150-ohm pullups and a bare STM32 cannot really drive those lines properly. Some F7 models have the output buffers. And of course plenty of people write 5.25 disks just fine with them.

Deksor, 2021-03-27:
I see. I guess I'll have to get an F7 model then.

keirf, 2021-03-27:
Ant Goffart has F7 Plus on sellmyretro. I will have a cheap new model (F7 v3) available in a couple of weeks. Those are your most cost effective options.

Deksor, 2021-03-27:
I think I'll buy a F7 v3

keirf, 2021-03-28:
I'll put you on the list.

keirf, 2021-04-06:
I have F7 v3 available now. Email me at keir.xen@gmail.com

Deksor, 2021-05-01:
I sent you a message 12 days ago (yes I'm a bit late ...) but didn't receive an answer yet, did it end up in your spam ?

keirf, 2021-05-01:
Sorry lost track. Replied now!

Deksor, 2021-05-14:
Thank you for sending me this !
Unfortunately, it did not solve any of the previously stated problems. So the mystery goes on. Most 5"1/4 drives still can't write, most can't read properly, using 3.5" drives do work however (like with the F1).

keirf, 2021-05-14:
Have you tried separately powering the drive (connect to its own 12v/5v power supply instead of the GW power connector)?

Most 5.25 drives need 12v which GW doesn't supply.

Deksor, 2021-05-14:
Yes, I used an AT power supply. A bit overkill for the task but it should work. I also added a hdd to put some load on the PSU to make sure it doesn't go out of specs.

keirf, 2021-05-14:
So let's say you try to read a known good disk with known good drive and known good cable. What happens?

Deksor, 2021-05-14:
Well it depends of the disk's density (DD or HD) and the drive used. But each drive has a specific pattern with the greaseweazle. However they work in PCs.

I'll do some testings tomorrow and will send you a screenshot of the data read from each disk in HxCFloppyEmulator

closed 2021-12-31 by keirf