keirf/greaseweazle #82: Read sometimes starts from nonzero track

philpem, 2021-02-22:
I'm using Greaseweazle to read 3.5in floppies with a Sony MPF520H drive.

Occasionally I'm finding that Greaseweazle doesn't seem to correctly step back to track zero. The resulting image is very weird:

* Track 14
* Track 13
* ...
* Track 0
* Track 1
* Track 2
* ...
* Track 63

When I re-read the disc, the image started at track 5, stepped back to 0, then stepped up to 77. On the third re-read, I got a complete image from track 0 to 81.

I've also seen this behaviour on a Sony MPF920E.

It seems most prevalent when the preceding read operation only covers a subset of the drive's tracks, e.g. 0 to 10.

keirf, 2021-02-22:
That's dubious. Never heard of it before. Could trk0 or dir signals be iffy?

Not getting to track 0 on demand is weird. Stepping out and then bouncing off track 0 and stepping inward is weird.

philpem, 2021-02-22:
I don't understand it either. I've attached one of the iffy image files for you to examine. Drop it into HxCFloppyEmulator, and track 0 is as described.

[Fabis_EasyClip_try4_MPF920_BAD.scp.zip](https://github.com/keirf/Greaseweazle/files/6024705/Fabis_EasyClip_try4_MPF920_BAD.scp.zip)

I'll check the wiring on the Track Zero pin.

philpem, 2021-02-22:
Track Zero and Direction wiring is fine - none of the resistor packs are shorted either.

philpem, 2021-02-22:
Ahhh, okay, I can see what's going on here. I had a floppy drive failure and hot-swapped to another drive.

Greaseweazle only does a full seek-to-zero recalibrate if `unit->initialised` is `false`. So swapping the drive is what broke this: the replacement drive wasn't seeked to the same track, and Greaseweazle lost track (pardon the slight pun).

I think I was expecting to behave like a DiscFerret and recalibrate the drive every time a read operation was started.
The DF also does a check for a bad track-zero sensor:

- seek to zero using TRACK0  (error "track 0 not reached" if TRACK0 didn't go active after e.g. 85 steps)
- seek +1 track and check TRACK0 is now inactive  (error "track 0 sensor stuck active" if TRACK0 didn't go inactive)
- seek back to zero and check TRACK0 went active again  (error "track 0 sensor misaligned or faulty" if TRACK0 didn't go active)

What can I say, I'm a pessimist :)

keirf, 2021-02-22:
Yes, Greaseweazle is a bit stateful. You can `gw reset` after a hot swap to reset everything.

keirf, 2021-02-22:
I may add `gw test` with some subcommands off that to do things like checking signals, run through an alignment test with a pre-written disk, etc etc...

closed 2021-02-26 by keirf