keirf/greaseweazle #87: Support TG43 on user output (for writing to 8" drives)

wohali, 2021-03-05:
This should be easy; a command line flag that sets 8" write mode would simply re-use `DENSEL`. This is already on pin 2, so an adapter would straight-through connect the 34- and 50-pin connectors on pin 2.

On 8" read mode, the line should be disabled.

keirf, 2021-03-05:
So this would be a flag to gw write which sets pin 2 low for tracks 43 and higher? (or is it higher than 43 but not 43 itself?)

wohali, 2021-03-05:
See section 7.13 (page 46) of https://hxc2001.com/download/datasheet/floppy/thirdparty/Shugart_Associates/shugart%20SA850%20OEM%20Manual%201980.pdf :

> When the interface signal is activated to a logical zero level, the lower value of the write current is selected for writing on tracks 43 through 76.

Note that per the document this same signal line may optionally be used on read as well as an active read filter:

> When the interface signal is activated to a logical zero level, the read signal is passed through an active filter network for reading tracks 60 through 76. Performance will improve when reading a diskette that has been recorded without write precompensation.
>
> To control the active read filter from the interface, short trace "IW" (to connect to the interface), remove jumpers AF and NF. When the interface is activated to a logical zero level, read compensation is selected as if AF were added. When the interface is activated to a +5V level, read compensation is selected as if NF were added.
>
> For optimum performance for FM or MFM encoding, jumper trace AF, for M²FM encoding jumper trace NF.

So this may still be useful in its standard sense for reading on an 8" drive. Of course this is only describing the SA850, I'm not sure how other 8" drives interpret this signal.

keirf, 2021-03-08:
I'm happy to add a flag, please just explain *exactly* what it should do and on what operations.

wohali, 2021-03-08:
Great, I've got parts for an F7 Lightning Plus coming in, once that's assembled and tested, and I experiment a little bit, I'll let you know.

closed 2021-12-31 by keirf

misterblack1, 2022-06-25:
Basically, we'd need two things to optionally be enabled to support 8" disks:

When writing -->

### When tracks 43 or greater are being written, a signal needs to go low which goes to the TG43 pin on the drive. There is no specific pin on the 34 pin connector to support this signal.

### Most adapters that convert from 34 pin shugart to 50 pin allow you to use the DENSEL signal to drive TG43 on the drive. So if you just keep DENSEL high for tracks 0-42 and then make it low for 43-255. (Or whatever the max track number is.)

Also, another options for reading which some drives can benefit from. When reading -->

### Keep DENSEL high for tracks 0-59 and bring DENSEL low for tracks 60 to 255. This is to help read disks not written with pre-compensation originally. (Drive that support this will turn on an extra filter.)

Shugart 850/851 use these signals. 
Shugart 800/801 do not use these signals.

keirf, 2022-06-25:
I will say you can do this already by splitting your read or write into two sections and invoking `gw pin set 2 L` (or H) in between.

misterblack1, 2022-06-25:
Indeed -- would be doable, although in the case of splitting up the read or write process, one could simply ground the pin on the drive directly without running an extra wire.

Thought it might not be too difficult to do in software to make the process simpler. 

Thanks

keirf, 2022-06-25:
Is TG43 the main one to support, for writes only?

sweproj, 2023-08-27:
Hi

I have implement TG43 on my board. 
Beware im not a developer so please dont kill me for bad code
My fork is here https://git.sweproj.com/sweproj.com/greaseweazle_f7_lightning_usb_pd
And the hw design here
https://git.sweproj.com/sweproj.com/greaseweazle_f7_lightning_usb_pd

Maybe hor somebody else to implement it by them self