keirf/greaseweazle #95: Floppy support

toncho11, 2021-04-03:
Hi,

Is this a USB floppy adaper? 

Is it both read and write?

Does it support 5.25 floppy disks?

keirf, 2021-04-03:
It is a USB floppy adapter but with a custom protocol and custom low-level read/write software supplied. It is not a "USB drive" in the sense you can connect a floppy drive and have it show up in Windows File Explorer (for example).

toncho11, 2021-04-05:
Please update the project description to avoid answering people like me :)
Now I get it. Great project!!! As a retro enthusiast I would like to write disk images directly from my Windows 10 PC to a 5.25 floppy. Actually you can give this as an example in the description.

closed 2021-04-05 by toncho11